platformio/platformio-vscode-ide #3227: Start PIO Home Server

lkdhruw, 2022-06-25:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect ETIMEDOUT 127.0.0.1:8010
	at c:\Users\lkdhr\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:9932
	at processTicksAndRejections (node:internal/process/task_queues:96:5)
  ```

ivankravets, 2022-06-27:
Have you resolved this issue?

JeremyPorquez, 2022-07-10:
@ivankravets
Had the same experience with VS Code. So I tried the CLI route. Installed pio via python (I have Python 3.10 on Mac OS 12.3.1). Symlinked the executables to /usr/local/bin. Then ran `$ pio home` but still could not connect to internet. I then resorted to `$ sudo pio home` and then it worked.

Jeremys-MacBook-Pro-2018:Library jeremyporquez$ pio home
Please wait while upgrading PlatformIO...
Error: You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.
Jeremys-MacBook-Pro-2018:Library jeremyporquez$ sudo pio home
Please wait while upgrading PlatformIO...
Tool Manager: Updating tool-scons @ 4.40300.1
Tool Manager: Removing tool-scons @ 4.40300.1
Tool Manager: tool-scons@4.40300.1 has been removed!
Tool Manager: Installing platformio/tool-scons @ 4.40300.220706
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40300.220706 has been installed!
PlatformIO has been successfully upgraded to 6.1.0!

closed 2022-09-20 by lkdhruw