platformio/platformio-vscode-ide #3229: Prevent users from using wrong build / debug buttons

maxgerhardt, 2022-06-26:
Per [forum conversation](https://community.platformio.org/t/arduino-h-no-such-file-or-directory/28474/16?u=maxgerhardt).

In VSCode, there is a "Play" button in the upper right corner of the opened code file, that will lead to VSCode asking to generate a new compiler / build configuration in accordance to detected global compilers:

[image]

[image]

Users then e.g. select a global avr-gcc installation and try to "build" it, which will of course immediately fail since none of the compiler flags and include paths weren't set up and the build doesn't go through PlatformIO

[image]

So they come to the conclusion that PlatformIO is broken and try to reinstall it, with no effect of course.

Can we somehow prevent the usage of these buttons from these extension?

The documentation ([1](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks), [2](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-toolbar), [3](https://docs.platformio.org/en/latest/integration/ide/vscode.html#debugging)) is already very clear on how to do a build and debug, but users who don't visit these pages and go straight for the kill might end up in the above situation.

ivankravets, 2022-06-28:
@sean-mcmanus, do you have any ideas on how could we disable these buttons for the PlatformIO projects? Maybe, extra options to `c_cpp_properties.json`?

sean-mcmanus, 2022-06-28:
@ivankravets Currently, those debug buttons are controlled by the C_Cpp.debugShortcut user settings -- you could theoretically overwrite this default in your extension with something like
```c
"configurationDefaults": {
  [c]: {
    "C_Cpp.debugShorcut": false
  }
}
```.
in your package.json file; however, that doesn't currently work because "C_Cpp.debugShortcut" doesn't have a "language-overridable" scope and VS Code doesn't appear to allow "application" scope settings to be set in "configurationDefaults". Otherwise, we'd need to change the setting in our extension to be settable by workspace or workspace folders settings and/or in c_cpp_properties.json. You could file an issue or PR at https://github.com/Microsoft/vscode-cpptools .

ivankravets, 2022-06-28:
@maxgerhardt could I ask you to file an issue at https://github.com/microsoft/vscode-cpptools/issues and put a link to this issue? It looks like a blocker issue and we need a fix ASAP. People install and uninstall the PlatformIO extension because face with this issue and think that it does not work.

sean-mcmanus, 2022-06-28:
I created a PR at https://github.com/microsoft/vscode-cpptools/pull/9514, but that would require your extension set
```json
"configurationDefaults": {
  [c]: {
    "C_Cpp.debugShorcut": false
  }
}
```
is that sufficient or did you still want some other change?

ivankravets, 2022-06-28:
Thanks so much for the quick fix! We appreciate this! 🙏

We will add your code to our “package.json”.

sean-mcmanus, 2022-06-29:
@ivankravets The language-override scope didn't work for some reason so we just changed it to resource scope and that works, so your extension needs to set:
```json
    "configurationDefaults": {
        "C_Cpp.debugShortcut": false
    }
```
in your package.json. This appears to work even in our released verion which has it set to application scope, but it shows an error/warning in the package.json unless the version of C/C++ with the scope change is installed. The advantage of the release scope is that it also allows per-workspace settings differences.

ivankravets, 2022-06-29:
See related commit https://github.com/platformio/platformio-vscode-ide/commit/e116c580b703c8749767a59fe42b097bff66aa4a

You can check it on your side:

1. Clone this repo
2. `yarn install`
3. Run without debug
4. Open any PlatformIO project.

Does it work for you?

sean-mcmanus, 2022-06-30:
@ivankravets yarn install gives "Error: Cannot find module 'platformio-vscode-ide\node_modules\platformio-vscode-debug\node_modules\@types\vscode\bin\install". Am I doing something wrong?

sean-mcmanus, 2022-06-30:
The "configurationDefaults" property needs to go under the "contributes" section. That is strange why VS Code doesn't show a warning. It should look like...

[image]

That works for me. FYI, the package.json warning should go away when we release 1.10.8.

sean-mcmanus, 2022-06-30:
https://github.com/microsoft/vscode-cpptools-api is a versioned API so there shouldn't be any breaking changes. It is used by CMake Tools, Makefile Tools, and a few other extensions.

The user would still get "include file not found" when the project is open, but they go away as soon as we get the configuration info from the configuration provider. Does that sound good enough or did you want a way to have IntelliSense disabled until the configuration data is provided? I recall the reason for the current behavior is that we don't have a way to know what we should wait for configuration info. We haven't received any user complaints about this so we haven't looked into improving it lately. You could file an issue at https://github.com/microsoft/vscode-cpptools/issues .

The API might solve other issues though. And maybe something could be added to the API to enable configuration providers to disable IntelliSense until configuration info is available.

I'm not familiar with extension enabling/disabling, but you could try that.

ivankravets, 2022-07-01:
Thanks, @sean-mcmanus! I've just opened a separate issue https://github.com/platformio/platformio-vscode-ide/issues/3232

closed 2022-07-01 by ivankravets

ivankravets, 2022-07-01:
Filed a separate issue https://github.com/microsoft/vscode-cpptools-api/issues/43