platformio/platformio-vscode-ide #3234: Start PIO Home Server

Hjj-wing, 2022-07-02:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Error: You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

	at c:\Users\何俊杰qww\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:9850
	at ChildProcess.c (c:\Users\何俊杰qww\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\何俊杰qww\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-02:
Could you test the links mentioned in this comment https://github.com/platformio/platformio-core-installer/issues/2246#issuecomment-1172891319

What is your speed for the 1st, 2nd, and 3rd links?

DikiFive, 2022-07-02:
it 's seem we have the same question

ivankravets, 2022-07-02:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2246

closed 2022-07-02 by ivankravets

officesteve, 2022-07-02:
Close this as complete