platformio/platformio-vscode-ide #3237: Start PIO Home Server

Ptycoy, 2022-07-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Error: HTTPSConnectionPool(host='api.registry.nm1.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/contrib-piohome (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1129)')))

	at c:\Users\adame\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:9850
	at ChildProcess.c (c:\Users\adame\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\adame\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

Ptycoy, 2022-07-05:
When i'm trying to open "version 3" of https://api.registry.nm1.platformio.org/ i get that : 
Something went wrong...
Cannot read properties of undefined (reading 'readFile')
Stack trace
Error: Cannot read properties of undefined (reading 'readFile')
    at t.BaseResolver.<anonymous> (https://redocly.github.io/redoc/redoc-demo.bundle.js:8:74579)
    at Generator.next (<anonymous>)
    at https://redocly.github.io/redoc/redoc-demo.bundle.js:8:72726
    at new Promise (<anonymous>)
    at n (https://redocly.github.io/redoc/redoc-demo.bundle.js:8:72471)
    at t.BaseResolver.loadExternalRef (https://redocly.github.io/redoc/redoc-demo.bundle.js:8:74361)
    at t.BaseResolver.<anonymous> (https://redocly.github.io/redoc/redoc-demo.bundle.js:8:75092)
    at Generator.next (<anonymous>)
    at https://redocly.github.io/redoc/redoc-demo.bundle.js:8:72726
    at new Promise (<anonymous>)

ReDoc Version: 2.0.0-rc.72
Commit: 2100b549

ivankravets, 2022-07-05:
Do you use a proxy at your company?

Ptycoy, 2022-07-05:
Yes...

ivankravets, 2022-07-06:
Have you configured it https://docs.platformio.org/en/latest/integration/ide/vscode.html#proxy-server-support ?

Ptycoy, 2022-07-06:
pio.exe was blocked by our system. 
After that I got troubles with SSL handshake for python and for platformio.org by the proxy. 
I saw on the web that it's a know issue, I let the IT find a way... 
Thank-you for this application, it's a very great tool !

closed 2022-09-20 by stale[bot]