platformio/platformio-vscode-ide #3239: PlatformIO installation fails

lolwheel, 2022-07-06:
Hi, I've been a happy Platformio IDE user for a while. It just stopped working yesterday and I failed fix or even reinstall it so here are the repro steps for the clean install that's failing for me.

1. Download the latest VirtualBox image of Ubuntu 22.04 (from osboxes.com in my case) and spin up a brand new virtual machine.
2. Make sure to update all packages to the latest version.
3. Install `build-essential` and `python3-venv` packages via `apt`.
4. Install `vscode` through official Microsoft `dpkg` repo, here is the full version reported by it:
```
Version: 1.68.1
Commit: 30d9c6cd9483b2cc586687151bcbcd635f373630
Date: 2022-06-15T02:58:26.441Z
Electron: 17.4.7
Chromium: 98.0.4758.141
Node.js: 16.13.0
V8: 9.8.177.13-electron.0
OS: Linux x64 5.15.0-40-generic
```
5. Install Platformio IDE from the extensions.
6. Watch the following errors in the VSCode console as the PIO IDE extension tries to bootstrap the installation:
```
console.ts:137 [Extension Host] Error: PlatformIO Core has not been installed yet!	at u.value (/home/osboxes/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:800244)	at async S.check (/home/osboxes/.vscode/extensions/platformio.platformio-ide-2.5.0/dist/extension.js:1:4227)	at async /home/osboxes/.vscode/extensions/platformio.platformio-ide-2.5.0/dist/extension.js:1:27230
I @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:473
_invokeHandler @ rpcProtocol.ts:458
_receiveRequest @ rpcProtocol.ts:374
_receiveOneMessage @ rpcProtocol.ts:296
(anonymous) @ rpcProtocol.ts:161
invoke @ event.ts:575
deliver @ event.ts:779
fire @ event.ts:740
fire @ ipc.net.ts:638
_receiveMessage @ ipc.net.ts:958
(anonymous) @ ipc.net.ts:831
invoke @ event.ts:575
deliver @ event.ts:779
fire @ event.ts:740
acceptChunk @ ipc.net.ts:382
(anonymous) @ ipc.net.ts:338
L @ ipc.net.ts:60
emit @ node:events:390
addChunk @ node:internal/streams/readable:315
readableAddChunk @ node:internal/streams/readable:289
Readable.push @ node:internal/streams/readable:228
onStreamRead @ node:internal/stream_base_commons:199
log.ts:313   ERR [Extension Host] (node:4079) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(Use `code --trace-deprecation ...` to show where the warning was created)
console.ts:137 [Extension Host] (node:4079) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.(Use `code --trace-deprecation ...` to show where the warning was created)
I @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:473
_invokeHandler @ rpcProtocol.ts:458
_receiveRequest @ rpcProtocol.ts:374
_receiveOneMessage @ rpcProtocol.ts:296
(anonymous) @ rpcProtocol.ts:161
invoke @ event.ts:575
deliver @ event.ts:779
fire @ event.ts:740
fire @ ipc.net.ts:638
_receiveMessage @ ipc.net.ts:958
(anonymous) @ ipc.net.ts:831
invoke @ event.ts:575
deliver @ event.ts:779
fire @ event.ts:740
acceptChunk @ ipc.net.ts:382
(anonymous) @ ipc.net.ts:338
L @ ipc.net.ts:60
emit @ node:events:390
addChunk @ node:internal/streams/readable:315
readableAddChunk @ node:internal/streams/readable:289
Readable.push @ node:internal/streams/readable:228
[... 122 lines cut ...]
L @ ipc.net.ts:60
emit @ node:events:390
addChunk @ node:internal/streams/readable:315
readableAddChunk @ node:internal/streams/readable:289
Readable.push @ node:internal/streams/readable:228
onStreamRead @ node:internal/stream_base_commons:199
mainThreadExtensionService.ts:80 RequestError: connect ECONNREFUSED 127.0.0.1:8036
	at ClientRequest.<anonymous> (/home/osboxes/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/got/dist/source/core/index.js:970:111)
	at Object.onceWrapper (node:events:510:26)
	at ClientRequest.emit (node:events:402:35)
	at ClientRequest.emit (node:domain:475:12)
	at ClientRequest.origin.emit (/home/osboxes/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/@szmarczak/http-timer/dist/source/index.js:43:20)
	at Socket.socketErrorListener (node:_http_client:447:9)
	at Socket.emit (node:events:390:28)
	at Socket.emit (node:domain:475:12)
	at emitErrorNT (node:internal/streams/destroy:157:8)
	at emitErrorCloseNT (node:internal/streams/destroy:122:3)
	at processTicksAndRejections (node:internal/process/task_queues:83:21)
	at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1161:16)
```

The last ```ECONNREFUSED``` error is logged many more times and the installation is stuck at that point, here's the screenshot of the window in that state:
[image]

Let me know if you need any more information, happy to provide.

ivankravets, 2022-07-06:
Please ignore `ECONNREFUSED` warnings. What is the end of the log? Could you close VSCode and try installation script https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#super-quick-macos-linux ? Does it work?

lolwheel, 2022-07-06:
Okay some more details:
```
$ netstat -tp
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp    82412      0 osboxes.lan:36626       cdn-185-199-109-1:https ESTABLISHED 12513/python3       
tcp        0      0 osboxes.lan:54848       banjo.canonical.co:http TIME_WAIT   -                   
tcp6       0      1 osboxes:34586           2606:50c0:8000::1:https SYN_SENT    12513/python3
```
the tcp6 is stuck in SYN_SENT while the regular IPv4 socket has a bunch of data stuck in its receiving buffer. I might have some ipv6 issues on the network. This fixed the PlatformIO installer script but the Build and Intellisense index rebuild still hang.

commit 03e8c3df (2022-07-01): Prevented users from using wrong Build/debug buttons // Resolve #3239
commit 06b3b95b (2022-07-28): Prevented users from using wrong Build/debug buttons // Resolve #3239
closed 2022-07-06 by lolwheel