platformio/platformio-vscode-ide #3240: Start PIO Home Server

michafroh, 2022-07-07:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.14393, x64

  %23 Exception
  ```
  Error: Error: 

	at c:\Users\dx5entkv\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:9850
	at ChildProcess.c (c:\Users\dx5entkv\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\dx5entkv\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-07:
Could you reproduce this issue after restarting  VSCode?

closed 2022-07-07 by ivankravets

michafroh, 2022-07-07:
yes, i can reproduce after restarting vscode, also after reboot the laptop.

michafroh, 2022-07-07:
i'm working with a companys laptop. I'm not really sure, which antivirus is installed. My collegue said, that we do not use any russion antivirus.

ivankravets, 2022-07-07:
Could you open the system terminal (cmd.exe) and type
```
python --version
```

Please provide an output.

michafroh, 2022-07-07:
i found a mail from our IT, that CrowdStrike Falcon is installed? Since Oct. 2021.

michafroh, 2022-07-07:
Python 3.8.1

michafroh, 2022-07-07:
C:\Users\dx5entkv\.platformio\penv\Lib\site-packages\scripts>python.exe get-platformio.py
Traceback (most recent call last):
  File "get-platformio.py", line 105, in <module>
    sys.exit(main())
  File "get-platformio.py", line 98, in main
    dst = download_file(MAIN_SCRIPT_URL, str(tmp_file.name))
  File "get-platformio.py", line 93, in download_file
    raise Exception("Could not download file '%s' to '%s' " % (url, dst))
Exception: Could not download file 'https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py' to 'C:\Users\dx5entkv\AppData\Local\Temp\tmpa0qh5sjr'

C:\Users\dx5entkv\.platformio\penv\Lib\site-packages\scripts>

ivankravets, 2022-07-07:
1. You used wrong script, you need to use `get-platformio.py` mentioned in the "Local Download (macOS / Linux / Windows)" section
2. It seems PlatformIO and PyPi are blocked by your network provider (firewall).

michafroh, 2022-07-07:
C:\Users\dx5entkv\Downloads>python.exe get-platformio.py
Installer version: 1.1.2
Platform: Windows-10
Python version: 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)]
Python path: C:\Program Files (x86)\Python38-32\python.exe
Creating a virtual environment at C:\Users\dx5entkv\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)'))': /simple/platformio/
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)'))': /simple/platformio/
  WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)'))': /simple/platformio/
  WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)'))': /simple/platformio/
  WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)'))': /simple/platformio/
  Could not fetch URL https://pypi.org/simple/platformio/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/platformio/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)'))) - skipping
  ERROR: Could not find a version that satisfies the requirement platformio (from versions: none)
ERROR: No matching distribution found for platformio
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)'))) - skipping
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\dx5entkv\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.

C:\Users\dx5entkv\Downloads>

michafroh, 2022-07-07:
ok, thanks for information, i'm going to contact him.