platformio/platformio-vscode-ide #3248: Start PIO Home Server

HeribertoRC, 2022-07-12:
Hello to everyone, do you have a solution for this problem? I have uninstalled and installed VSCode 4 times today and keeps happening 

  %23 Configuration

  VSCode: 1.69.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Error: 

	at c:\Users\z0000318\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:9850
	at ChildProcess.c (c:\Users\z0000318\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\z0000318\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-15:
Could you reproduce this issue after restarting VSCode?

HeribertoRC, 2022-07-15:
Yes, this issue is repeatable even when doing a restart of VSCode or when doing a clean reinstallation

closed 2022-09-27 by stale[bot]