platformio/platformio-vscode-ide #3259: ESP32 Upload fails (multiple reasons) workaround is restarting VS Code / PlatformIO

robomike, 2022-07-20:
Hi All, 

I find this a bit annoying when developing with platformIO for ESP32 and Arduino.

I am using:

Ubuntu 20.04
VS Code with PlatformIO
ESP32

Task Upload Fails:

Error:
File "/home/futuregen/.platformio/packages/tool-esptoolpy/esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration
*** [upload] Error 1
===================== [FAILED] Took 5.98 seconds =====================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

Additional Error, if I don't get the first one above:
A fatal error occurred: Failed to connect to ESP32: Serial data stream stopped: Possible serial noise or corruption.

Workaround:
Resart VS Code (close application and launch it again)
Run Task Upload - works with no issues.

My ini file:
[env:sparkfun_lora_gateway_1-channel]
platform = espressif32
board = sparkfun_lora_gateway_1-channel
framework = arduino
upload_port = /dev/ttyUSB0
monitor_speed = 9600
lib_deps = 
    mikem/RadioHead@^1.120
    #sandeepmistry/LoRa @ ^0.8.0
    sparkfun/SparkFun u-blox GNSS Arduino Library@^2.2.10

ivankravets, 2022-07-21:
Could you try to run uploading without VSCode?

```
~/.platformio/penv/bin/platformio run -d /path/to/project -t upload
```

Does it work?

robomike, 2022-07-21:
The root cause of the issue is that there was an open terminal using Monitor and PlatformIO does not close the open terminal.  You can close the active terminal and it resolves this issue.

ivankravets, 2022-07-21:
If you opened the serial monitor the first time without PlatormIO, we will not close it because we don't know that you opened it.

robomike, 2022-07-21:
@ivankravets I only work inside PlatformIO so it should know it is opened.

ivankravets, 2022-07-21:
Did you change this option https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-autocloseserialmonitor ?

robomike, 2022-07-21:
I have not changed any default options, where would I find that option to verify the setting?

ivankravets, 2022-07-21:
If opened the serial monitor as a separate task - we will not close it. Could test the next:

1. Close all serial monitors / tasks
2. Run Upload & Monitor
3. Repeat again.

Does it device monitor before uploading and reopening after?

robomike, 2022-07-21:
I just use the tasks.  If the serial monitor is open, the upload fails and the serial monitor succeeds.  Hence, no update.

ivankravets, 2022-07-21:
Yed, it should fail in this way. PlatformIO closes only the previous serial monitor that was opened by the previous upload task (using Upload & Monitor). If you open manually monitor and later use Upload & MOnitor - it will not close the active serial monitors because we don't know what developers actually want.

closed 2022-07-21 by ivankravets

NXTkoji, 2022-12-15:
I am using esp32-wroom-32 Doit DevKit and had a similar problem of:
A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
*** [upload] Error 2
I found out that I had D16 and D17 pins, which happens to be RXD2 and TXD2, connected to my relays. Once I changed them to other pins, the problem was solved.

hieult61, 2023-08-17:
> Hi All,
>
> I find this a bit annoying when developing with platformIO for ESP32 and Arduino.
>
> I am using:
>
> Ubuntu 20.04 VS Code with PlatformIO ESP32
>
> Task Upload Fails:
>
> Error: File [...]

I have the same issue. Actually i have only my project running VScode. Can you tell me what the other open terminal using monitor is ? Thank you