platformio/platformio-vscode-ide #3261: esp32-c3 An Error has occurred while mounting SPIFFS

cug-auto-zp, 2022-07-22:
I used vscode + platformIO + Arduino to code and upload esp32-c3. 
Now i am in trouble.
I want to used SPIFFS lib, so i found a web (https://randomnerdtutorials.com/esp32-vs-code-platformio-spiffs/) which teaches me how to the SPIffs lib of Arduino succeed upload the `.bin` in data file.
code: 
[image]

[image]

[image]

But the problem is by monitor : 

[image]

ivankravets, 2022-08-12:
Please report to https://community.platformio.org/

closed 2022-08-12 by ivankravets

cug-auto-zp, 2022-08-12:
Thank you send E-mail to me.l am busy in studying .lf l read it,l will reply you  the first time.