platformio/platformio-vscode-ide #3263: Installing does not work on M1 Mac

fellnerse, 2022-07-24:
The installation just does not continue and stops at "PlatformIO Installer: Installing PlatformIO Core".
I can install the core with "brew install platformio" and use it but dont know how to link it.

It seems to be a problem of pip, when running the "~/.platformio/.cache/tmp/get-platformio-1.1.2.py" command from hand this also just stops doing anything after printing:
```bash
Installer version: 1.1.2
Platform: macOS-12.4
Python version: 3.10.0 (default, Dec 21 2021, 13:39:52) [Clang 13.0.0 (clang-1300.0.29.30)]
Python path: ~/.pyenv/versions/3.10.0/bin/python
Creating a virtual environment at ~/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
```

Error from developer tools in VS code:

```
RequestError: connect ECONNREFUSED 127.0.0.1:8010
	at ClientRequest.<anonymous> (~.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/got/dist/source/core/index.js:970:111)
	at Object.onceWrapper (node:events:510:26)
	at ClientRequest.emit (node:events:402:35)
	at ClientRequest.emit (node:domain:475:12)
	at ClientRequest.origin.emit (~/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/@szmarczak/http-timer/dist/source/index.js:43:20)
	at Socket.socketErrorListener (node:_http_client:447:9)
	at Socket.emit (node:events:390:28)
	at Socket.emit (node:domain:475:12)
	at emitErrorNT (node:internal/streams/destroy:157:8)
	at emitErrorCloseNT (node:internal/streams/destroy:122:3)
	at processTicksAndRejections (node:internal/process/task_queues:83:21)
	at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1161:16)
```
and
```
~/.platformio/python3/bin/python3 Error: dyld[46976]: Library not loaded: /usr/local/opt/gettext/lib/libintl.8.dylib
  Referenced from: ~/.platformio/python3/bin/python3.9
  Reason: tried: '/usr/local/opt/gettext/lib/libintl.8.dylib' (no such file), '/usr/local/lib/libintl.8.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libintl.8.dylib' (no such file)

	at /Users/sebastianfellner/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5227
	at ChildProcess.c (/Users/sebastianfellner/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Socket.<anonymous> (node:internal/child_process:450:11)
	at Socket.emit (node:events:390:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
```

For the second one I tried to link this myself as you can see its the wrong architecture, but before it was not able to find it at all.

Please fix this, is not usable for me at all atm and can not isntall an old version (used to work for me)

jrmann100, 2022-07-24:
I'm also having issues on my M1. Are you using the Apple Silicon version of VSCode, or the universal binary?

fellnerse, 2022-07-25:
So I was able to install 2.4.2 of platformio

this is the information of vscode:
```
Version: 1.69.2
Commit: 3b889b090b5ad5793f524b5d1d39fda662b96a2a
Date: 2022-07-18T16:32:39.000Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Darwin arm64 21.5.0
```

jrmann100, 2022-07-25:
What helped for me was trying to either a) find the VSCode application in Finder and right-click > Get Info > Open using Rosetta or b) installing the "Universal" release at https://code.visualstudio.com/download. The Apple Silicon release of VSCode that `brew --cask` chose did not work for me, but I'm not sure if you got it through there.