platformio/platformio-vscode-ide #3272: Platform IO installation infinite loop

singletone, 2022-08-07:
Hello, Ivan,
Today I got the issue with 'Checking Platform installation' infinite loop (similar to the one described in #639). Following your recommendation, I have removed the folder "C:/Users/%Name%/.platformio/penv". This didn't help. Closing Python processes didn't help either. Attempts to re-install Platform IO ended up with "PlatformIO Installer: Installing PlatformIO Core" - the process got stuck. It looks like this is caused by a sudden update. Could you please help me out?

singletone, 2022-08-08:
I also tried to install PlatformIO Core manually. Downloaded the installer and run the following: get-platformio.py. The process stuck at this: 'Creating a virtual environment at C:\............\.platformio\penv'
Any suggestions?

singletone, 2022-08-10:
In the long run, I got it to work. 
- I re-installed Python (although I'm not sure that was the matter), 
- then I manually removed all references to platformio in the Windows 11 Registry (unfortunately, 'Uninstall' doesn't do this). 
- Also, I removed .platformio directory (in fact, just renamed it). 
- After that, I ran Install in VS Code. This time it was finished successfully.
I hope this will be helpful to someone facing the same problem.

closed 2022-08-10 by singletone

bodaxsd, 2023-03-02:
你需要重新加载代理服务器