platformio/platformio-vscode-ide #3278: PIO Home not showing up

TheMakerJojo, 2022-08-13:
Hello.
I have a problem using PlatformIO in codeserver. 
I installed the extension using the VSIX-File from the VS code marketplace.
Codeserver is running on an Raspberry Pi 4b with 2GB of RAM which is connected to my iPad Pro M1 via usb-c cable.
The connection is set up like in this tutorial: https://www.hardill.me.uk/wordpress/2019/11/02/pi4-usb-c-gadget/

The PIO Home won’t show up in codeserver. See in the picture below:
[image]

If i do „pio home --host 10.55.0.1“ to display PIO Home in my Webbrowser it works perfectly fine.
[image]
[image]

Please help me!

ivankravets, 2022-08-13:
Did you open http://10.55.0.1:8010?

TheMakerJojo, 2022-08-13:
When „platformio-ide.pioHomeServerHttpHost“ is set to „0.0.0.0“ or „10.55.0.1“ and i open „10.55.0.1“ in the browser, there is an message: „Protected PlatformIO Home session“.

TheMakerJojo, 2022-08-18:
In this video you can see the problem.

https://youtu.be/AOgNRvKYhoA