platformio/platformio-vscode-ide #3280: Optimize the installation experience

Osirisoo0O, 2022-08-15:
Error: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001BC135757C0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001BC13575730>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=Non...

ivankravets, 2022-08-15:
There is no access to the PyPi registry. Do you use proxy or custom PyPi index?

Osirisoo0O, 2022-08-16:
> There is no access to the PyPi registry. Do you use proxy or custom PyPi index?

Yes. But I find other way to fix it. Now it Works!
Anyway,thanks for your reply!

ivankravets, 2022-08-16:
Glad to hear from you. Happy coding with PlatformIO 🚀

closed 2022-08-16 by ivankravets