platformio/platformio-vscode-ide #3282: vscode -> PlatformIO -> ESP32

hzhh110, 2022-08-18:
[image]

I don't know what's going on, ubuntu install vscode -> PlatformIO -> ESP32, it has been stuck here and can't be successful

ivankravets, 2022-08-18:
Could you reproduce this issue after restarting of VSCode?

hzhh110, 2022-08-20:
> Could you reproduce this issue after restarting of VSCode?

Ubuntu 20.04.4 and Ubuntu 18.04.6 both fail to be installed. Ubuntu 20.04.4 python3-venv : Depends: python3 (= 3.6.7-1~18.04) but 3.8.2-0ubuntu2 is to be installed
[image]