platformio/platformio-vscode-ide #3287: http.proxy in vscode is not set for build task

yyjdelete, 2022-08-24:
https://github.com/platformio/platformio-vscode-ide/blob/e116c580b703c8749767a59fe42b097bff66aa4a/src/main.js#L126-L150

https://github.com/platformio/platformio-vscode-ide/blob/2ed0070f71c27a1251cb3a1bdd97ede1006875c4/src/project/tasks.js#L111-L116

closed 2022-10-31 by ivankravets

ivankravets, 2022-10-31:
Thanks for the report! Please update the extension to the v2.5.5

AndreyNazarenko62, 2022-11-23:
Установил VSCODE для работы с esp32-wroom-32 devkit 1, в качестве расширения PlatformIO IDE, один раз скомпилировал пример со светодиодом, прошивка прошла, все заработало. Второй раз добавил в прошивку вывод числа в незанятый UART, тоже в норме, на осциллографе видно импульс. Потом стала возникать систематическая ошибка при сборке, потом пропала <Arduino.h> и наконец исчезла возможность запустить сборку и отладку.
Подскажите в каком направлении копать? Переустанавливал дважды - толку нетю

ivankravets, 2022-11-23:
@AndreyNazarenko62, this issue was related to the HTTP Proxy. If you don't use a Proxy, please forward your issue to https://community.platformio.org/ and our awesome community will help you. Thanks!

commit 881caddd (2022-10-31): Fixed an issue with passing process.env to the child processes // Resolve #3287