platformio/platformio-vscode-ide #3299: VSCode Update 1.71 - Project Tasks Disabled

markxoe, 2022-09-02:
Hi!
The newest VSCode update caused an bug which leads to all the tasks in the "Project Tasks" tab being disabled:

[image]

(I have tried reinstalling PlatformIO IDE & Core and downgrading PlatformIO IDE)
Looking at VSCode's release notes I notice this: https://code.visualstudio.com/updates/v1_71#_disabled-tree-items

Idk if I am the only one having this problem.

Thanks!

markxoe, 2022-09-02:
Okay so this is because of these lines:
https://github.com/platformio/platformio-vscode-ide/blob/e30d96b470878fcb0cebf6ee6b217db33c7f038c/package.json#L96-L99

zekageri, 2022-09-02:
Did you solve it?

markxoe, 2022-09-02:
Well, it is still usable but its far more convenient to use the project tasks
I have some ideas to fix it but no idea what the best is 😄

GhoweVege, 2022-09-02:
Having this problem too. It's made my workflow a lot tougher to have to try to use the command line every time I need to do something. How can this be fixed?

markxoe, 2022-09-02:
Well, the probably worst fix would be to change `"enablement": "false"` to `"enablement": "true"`
The right way would probably be to have an context variable e.g. `platform-ide.canRunTasks` and to set it in the code accordingly

I could do an PR as an temporary fix for this using the first method and have a look at the better solution as soon as I have time

markxoe, 2022-09-02:
Plot twist: I've just tried this and I guess there are some more changes to the vscode api as clicking the button causes this window to open:

[image]

markxoe, 2022-09-02:
I've made an PR, it works on my machine. Maybe someone can test it? (Clone my repo, run `git checkout fix/project_tasks`, run `npm i`, open the clone in vscode and press f5 to test it)

closed 2022-09-03 by ivankravets

ivankravets, 2022-09-03:
[PlatformIO IDE v2.5.3](https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.5.3) for VSCode is out. We added a workaround on our side. It is unclear how to hide private/internal commands from user space with the latest VSCode.

commit f7db74eb (2022-09-03): Fixed an issue when Project Tasks are disabled in the VSCode v1.71 // Resolve #3299