platformio/platformio-vscode-ide #3335: remote upload fails on missing snap

arpaterson, 2022-10-04:
I installed platform io via vscode extension manager on both client (windows11) and host (jetson orin, ubuntu20.04).

I have gotten remote agent to start (problems with rustc, cargo, cryptography) after logging in, generating an auth token, setting up udev rules, and starting pio remote agent with the auth token.

I can see the remote agent listed, and the devices on it.

But remote agent fails to upload with the following error about `/snap/bin/platformio`.

```
Uploading firmware remotely
Upon execvpe platformio ['platformio', '--force', 'run', '-d', '/home/sacvp-user/.platformio/remote/projects/servoctl-f7a77c20badfed8d3fde0b75f5ab5dc51d078e4f', '-e', 'Adafruit Feather M4 Express', '-t', 'upload', '-t', 'nobuild', '--disable-auto-clean'] in environment id 281473791027904
:Traceback (most recent call last):
  File "/home/sacvp-user/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 397, in _fork
    self._execChild(path, uid, gid, executable, args, environment)
  File "/home/sacvp-user/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 468, in _execChild
    os.execvpe(executable, args, environment)
  File "/usr/lib/python3.8/os.py", line 577, in execvpe
    _execvpe(file, args, env)
  File "/usr/lib/python3.8/os.py", line 610, in _execvpe
    raise last_exc
  File "/usr/lib/python3.8/os.py", line 601, in _execvpe
    exec_func(fullname, *argrest)
FileNotFoundError: [Errno 2] No such file or directory: b'/snap/bin/platformio'

 *  The terminal process "C:\Users\arpat\.platformio\penv\Scripts\platformio.exe 'remote', 'run', '--target', 'upload', '--environment', 'Adafruit Feather M4 Express'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

To be clear, platformio was not installed via apt or snap.

On the host jetson:
```
$ which platformio
/home/sacvp-user/.platformio/penv/bin/platformio
```

Working via Remote SSH extension is also broken due to the problems with platformio's handling of file paths across windows/unix platforms...

closed 2022-12-16 by stale[bot]

diaclectic, 2023-01-04:
We had this same issue, and found a workaround:
`sudo mkdir /snap/bin && sudo ln -s /home/<username>/.platformio/penv/bin/platformio /snap/bin/platformio`

arpaterson, 2023-01-31:
This should be reopened, as the user install environment should be searched as well.

KenwoodFox, 2023-05-15:
This is what worked for me on KDE Neon
```
sudo mkdir -p /snap/bin && sudo ln -s /home/$(echo $USER)/.platformio/penv/bin/platformio /snap/bin/platformio
```