platformio/platformio-vscode-ide #3346: Does not work in VS Code 1.72.3 or 1.73.0 insiders -- "Buffer() is deprecated due to security and usability issues"

scottmudge, 2022-10-14:
Whenever PlatformIO is enabled in VS Code, I get the following error in the Developer Tools console, and the extension fails to load:

`DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.`

I've looked thoroughly but cannot fix it.

logana60, 2022-10-29:
Same issue, platformio previously working fine. 

Currently tried:
- restarting vscode
- deleting .platformio caches
- reinstalling c++ tools and platformio extensions
- installing insider versions of extensions
- reinstalling vscode

After installing the insiders build of vscode, platformio was able to launch for the first launch only. Subsequent IDE launches are now failing again.

scottmudge, 2022-10-29:
> After installing the insiders build of vscode, platformio was able to launch for the first launch only. Subsequent IDE launches are now failing again.

I noticed the exact same issue.

I will say, if I open a PlatformIO project, the buttons on the **bottom** of the UI (build, clean, etc) work fine. But the quick-access button on the left (the PIO icon) does not appear.

ivankravets, 2022-11-03:
Please update VSCode + PlatformIO IDE extension to the latest version. Remove `HOME_DIR/.platformio` folder and restart VSCode. Does it work now?

scottmudge, 2022-11-03:
Using VS Code Insiders v1.74.0 and PIO latest, clean `.platformio` folder. 

Activity button is still missing, though PIO seems to "work" beyond that. Deprecation warning still shows up, but even when PIO is disabled, so I think it's another extension still producing the error.

Any reason why the Activity button would disappear?

[image]

ivankravets, 2022-11-15:
@scottmudge , sorry for the late response. Have you resolved this issue? Do you see PlatformIO Activity button?

scottmudge, 2022-11-15:
No it's still not visible, on the latest VSCode Insider build and w/ the latest version of PlatformIO. PIO works otherwise, but the activity button is still missing.

ivankravets, 2022-11-15:
@scottmudge, could you try the latest stable version of VSCode? Does it work as expected?

scottmudge, 2022-11-15:
Stable version is the same, PIO works but no Activity bar button.

ivankravets, 2022-11-15:
I don't think this issue is linked to PlatformIO. Other extensions might add problems here. Could I ask you to uninstall stable VSCode + [Clean uninstall](https://code.visualstudio.com/docs/setup/uninstall#_clean-uninstall)? Then, please install clean VSCode and PlatformIO extensions. Please don't install other extensions. 

Do we have progress here?

logana60, 2022-11-15:
Personally, the clean install and then only installing c++ extension needed by platformio and platformio did not work for myself.

ivankravets, 2022-11-15:
@logana60 what is your VSCode and Windows versions?

scottmudge, 2022-11-15:
Same, did not work for me either. Completely clearing all cache/data, reinstalling, and installing only PIO causes the activity button to appear briefly on first install, then it disappears again.

Even if it was another extension (which it appears not to be), shouldn't PIO be encapsulated enough not to be affected by another extension?

ivankravets, 2022-11-18:
Could someone provide a remote session to debug this issue under your control? Please email me at me@ikravets.com for the details.

ivankravets, 2022-11-18:
@logana60, could you provide a remote session to debug this issue? Please ping me at me@ikravets.com

ivankravets, 2022-11-18:
We got more details about this problem, see the screenshot below (@logana60 , thanks for the provided `cpuprofile`):

[image]

--

@logana60, @scottmudge, have you set any limitations to the `process.env`? The screenshot above demonstrates that the PlatformIO IDE extension does not have access to the `process.env` and just freezes here.

ivankravets, 2022-11-18:
https://github.com/platformio/platformio-node-helpers/blob/master/src/proc.js#L37

Do you use proxy?

logana60, 2022-11-18:
> https://github.com/platformio/platformio-node-helpers/blob/master/src/proc.js#L37
> 
> Do you use proxy?

I am not familiar with process.env I haven't adjusted any node.js that I know of. 

I can try and add a few prints to that code and see where it gets stuck.

In the meantime, I gathered the process.env output from the vscode dev tools console
`{
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "APPDATA": "C:\\Users\\Logan\\AppData\\Roaming",
    "CHROME_CRASHPAD_PIPE_NAME": "\\\\.\\pipe\\crashpad_67252_NPDYSBMKOGUISNYG",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "COMPUTERNAME": "PC4350",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "HOMEDRIVE": "C:",
    "HOMEPATH": "\\Users\\Logan",
    "LOCALAPPDATA": "C:\\Users\\Logan\\AppData\\Local",
    "LOGONSERVER": "\\\\PC4350",
    "NUMBER_OF_PROCESSORS": "12",
    "OculusBase": "C:\\Program Files\\Oculus\\",
    "OneDrive": "C:\\Users\\Logan\\OneDrive",
    "OneDriveConsumer": "C:\\Users\\Logan\\OneDrive",
    "ORIGINAL_XDG_CURRENT_DESKTOP": "undefined",
    "OS": "Windows_NT",
    "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\Program Files\\Python310\\Scripts\\;C:\\Program Files\\Python310\\;C:\\Users\\Logan\\Downloads\\ffmpeg-2022-10-17-git-3bd0bf76fb-full_build\\bin;%PATH%;C:\\Program Files\\PowerShell\\7\\;C:\\Program Files\\dotnet\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Program Files\\LLVM\\bin;\"C:\\Users\\Logan\\.platformio\\penv\\Scripts;\";C:\\Users\\Logan\\AppData\\Local\\Programs\\Python\\Python311\\Scripts\\;C:\\Users\\Logan\\AppData\\Local\\Programs\\Python\\Python311\\;C:\\Users\\Logan\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\Logan\\AppData\\Local\\GitHubDesktop\\bin;C:\\Program Files (x86)\\Nmap;C:\\Users\\Logan\\.platformio\\penv\\Scripts;C:\\Users\\Logan\\AppData\\Local\\atom\\bin;C:\\msys64;C:\\MinGW;",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW",
    "POWERSHELL_DISTRIBUTION_CHANNEL": "MSI:Windows 10 Education",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
    "PROCESSOR_LEVEL": "23",
    "PROCESSOR_REVISION": "0802",
    "ProgramData": "C:\\ProgramData",
    "ProgramFiles": "C:\\Program Files",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "ProgramW6432": "C:\\Program Files",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules;C:\\Program Files (x86)\\Microsoft SQL Server\\120\\Tools\\PowerShell\\Modules\\",
    "PUBLIC": "C:\\Users\\Public",
    "SESSIONNAME": "Console",
    "SystemDrive": "C:",
    "SystemRoot": "C:\\WINDOWS",
    "TEMP": "C:\\Users\\Logan\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Logan\\AppData\\Local\\Temp",
    "USERDOMAIN": "PC4350",
    "USERDOMAIN_ROAMINGPROFILE": "PC4350",
    "USERNAME": "Logan",
    "USERPROFILE": "C:\\Users\\Logan",
    "VSCODE_CODE_CACHE_PATH": "C:\\Users\\Logan\\AppData\\Roaming\\Code\\CachedData\\6261075646f055b99068d3688932416f2346dd3b",
    "VSCODE_CWD": "C:\\Program Files\\Microsoft VS Code",
    "VSCODE_IPC_HOOK": "\\\\.\\pipe\\ae0439ae23f71eb64e7af01730697561-1.73.1-main-sock",
    "VSCODE_NLS_CONFIG": "{\"locale\":\"en-us\",\"availableLanguages\":{},\"_languagePackSupport\":true}",
    "VSCODE_PID": "67252",
    "windir": "C:\\WINDOWS"
}`

ivankravets, 2022-12-26:
Please re-test with the pre-release version. See the instructions below:

[image]

Does it work for you?

closed 2022-12-26 by ivankravets

logana60, 2022-12-27:
Yeah v2.99.2 and 2.99.3 works for me now. Thank you!

logana60, 2022-12-28:
Update: after multiple days and restarts with installs of the Arduino and Teensy platforms had zero issues. After installing ESP-IDF everything was working fine but upon relaunching vscode I am back to the previous issue of "initializing PlatformIO Core". This very likely could have caused my previous issue as well.

ivankravets, 2022-12-28:
Did you mean you installed ESP-IDF extension?

logana60, 2022-12-28:
yeah from inside of platformio. Tried uninstalling vscode, deleting every file that was titled/contained esp/pio/platformio/code/vscode and then reinstalling. still the same issue.

ivankravets, 2022-12-28:
Could you try to close the workspace/folder in VSCode, and restart VSCode? Now, open the PlatformIO project. Does it work now?

logana60, 2022-12-28:
1. Tried closing folder with ctrl+k F, no folders open
2. Closed vscode
3. reopened vscode, code opened without any folders open
4. Platformio still stuck on "initializing Platformio Core..."
5. Opened previous arduino(non ESP) project (no change)