platformio/platformio-vscode-ide #3351: Change toolbar

ygoe, 2022-10-22:
There is a usability issue with the toolbar of Platform IO in VSCode. I just wanted to upload my application to a device. Since my upload circuitry isn't stable yet, I tried it several times. But each time, the toolbar buttons shift to the side, left and right and back. Of course I accidently clicked on the wrong button because they were moved away under the mouse pointer as I was clicking the Upload button. So I hit the Clean button and now have to wait a couple of minutes again for a useless full ESP-IDF build.

This is a terrible design of a toolbar UI. The buttons in a toolbar shouldn't move anywhere, especially while clicking one. But it seems like they often do in VSCode. So maybe that tiny bottom toolbar isn't a good solution for the main actions of Platform IO. I suggest putting them somewhere else. I don't know too much about how this can be done in VSCode. But in most other IDEs these buttons are somewhere near the top of the window, and they remain in place.

I also suggest adding a button to upload without build. When I just want to re-upload some application to the device, I don't need a build step. That takes additional time looking for changes and whatever. Many seconds waiting can be saved if I could tell the IDE to just upload the most recent build, not try to build again when nothing has changed.

ivankravets, 2022-12-22:
Resolved via https://github.com/platformio/platformio-vscode-ide/commit/1e449aacb7ee0ea0d46ccde71acb23cb8c89b150

ivankravets, 2022-12-22:
Please re-test with the pre-release version. See instructions below:

[image]

closed 2022-12-22 by ivankravets