platformio/platformio-vscode-ide #3364: Command-A Behaves highlights screen in a seemingly useless and inescapable way

owenfi, 2022-11-05:
Type: <b>Bug</b>

Command-A keyboard shortcut on PlatformIO Home screen highlights the whole screen in a useless way that can't be reverted by hitting escape.

I use Command-A frequently (more often when system text input behaviors such as double click to highlight a word aren't respected) and in Platform IO this results in really weird behavior of coloring the entire view frame in a light gray box that I can find no way to get rid of.

Steps to reproduce:
Go to PlatformIO home view and click anywhere inside the view (not on any button or anything)
Press Command-A
See a gray box cover the view frame.

This same thing happens if I go to New Project, try to double/triple click the title (to highlight the entire word) and notice that it isn't highlighting. I observe the text cursor is blinking in the field. I hit Command-A to "select all" which is a very important Mac keyboard shortcut. Instead of highlighting the text (the indisputably proper platform behavior) a weird gray box is drawn. I can't find any way to make the box go away without losing my place, so I close the tab and open a fresh view which will no doubt be covered in the same gray box the next time a text field is broken and I instinctively hit Command-A again. It feels like this should at least be revertible with the Escape key, but that closes the dialog but doesn't hide the weird gray box.

Extension version: 2.5.5
VS Code version: Code 1.73.0 (8fa188b2b301d36553cbc9ce1b0a146ccb93351f, 2022-11-01T15:33:40.745Z)
OS version: Darwin arm64 21.2.0
Modes:
Sandboxed: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Apple M1 Max (10 x 24)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: disabled_off<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>skia_renderer: enabled_on<br>video_decode: enabled<br>video_encode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled<br>webgpu: disabled_off|
|Load (avg)|2, 3, 3|
|Memory (System)|64.00GB (17.98GB free)|
|Process Argv|--crash-reporter-id 0757656b-c7cc-4989-bd7b-234ec8586f4a|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vsreu685:30147344
python383cf:30185419
vspor879:30202332
vspor708:30202333
vspor363:30204092
vswsl492cf:30256860
vstes627:30244334
vslsvsres303:30308271
pythonvspyl392:30443607
vserr242:30382549
pythontb:30283811
vsjup518:30340749
pythonptprofiler:30281270
vshan820:30294714
vstes263:30335439
pythondataviewer:30285071
vscod805cf:30301675
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
cmake_vspar411:30581797
vsaa593:30376534
pythonvs932:30410667
cppdebug:30492333
vsclangdc:30486549
c4g48928:30535728
dsvsc012:30540252
azure-dev_surveyone:30548225
pyindex848:30577860
nodejswelcome1cf:30587006
fc301958:30595537
2e4cg342:30602488
gswce2:30602635
3d0df643:30602300
dbltrim-control:30603952

```

</details>

owenfi, 2022-12-14:
I have a hunch it's related to this change, will investigate further once I get time to figure out how to run a local copy of the plugin: https://github.com/platformio/platformio-vscode-ide/commit/a1acff8436b4e12dd947141afceb0986e9c9528b

ivankravets, 2023-02-01:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3119

PR #3451 (2023-02-24): Fix select all, undo, and redo on macOS
`selectAll`, `undo`, and `redo` are not supported events in `document.addEventListener` (https://developer.mozilla.org/en-US/docs/Web/Events).

To handle select all, we can detect a `selectStart` event, which only fires if something outside the iframe is selected. This is only achievable if select all is used as the iframe takes up the entire window. If so, then we execute the command and prevent the incorrect selection from occurring.

Undo/redo events do not exist (https://stackoverflow.com/a/15516286), so we can instead manually listen for Cmd+Z/Cmd+S [... cut]
closed 2023-02-01 by ivankravets