platformio/platformio-vscode-ide #3365: Syntax error in autogenerated .vscode/c_cpp_properties.json when using Arduino on STM32

ttytyper, 2022-11-08:
The variants directories contain parentheses. These are escaped with backslashes in c_cpp_properties.json. It seems that this causes the C/C++ extension to fail parsing the file.

Steps to reproduce:

1. Create PlatformIO project in VSCode
   Board: ST Nucleo L496ZG (possibly affects all STM32 boards, but this is what I use)
   Framework: Arduino
2. Open project folder if it isn't opened automatically
3. The following error message pops up in lower right corner of VSCode, even before editing or opening any files:

```
Failed to parse "/home/jesper/tmp/c_cpp_properties-syntax-error-test/.vscode/c_cpp_properties.json": Unexpected token ( in JSON at position 85

Source: C/C++ (Extension)
```
The following line shows up twice in c_cpp_properties.json, underlined with squiggles that assume means syntax error:

```
"/home/jesper/.platformio/packages/framework-arduinoststm32/variants/STM32L4xx/L496Z\(E-G\)T_L4A6ZGT",
```
Note the parentheses escaped with backslash.

I have attached the full json file as well as platformio.ini.
[c_cpp_properties.json.txt](https://github.com/platformio/platformio-vscode-ide/files/9959819/c_cpp_properties.json.txt)
[platformio.ini.txt](https://github.com/platformio/platformio-vscode-ide/files/9959823/platformio.ini.txt)

May be related to issue #1524. Error from OP mentions includePath (which is one of the JSON sections with syntax error), and another commenter specifically has problems with IntelliSense when using Arduino on STM32 boards.

sebsub321, 2022-11-24:
"/home/jesper/.platformio/packages/framework-arduinoststm32/variants/STM32L4xx/L496Z\(E-G\)T_L4A6ZGT",

ivankravets, 2022-12-26:
Should be fixed in the latest release.

closed 2022-12-26 by ivankravets

manuelbl, 2023-01-12:
In was release is this fix included? I'm hitting the same problem.

ivankravets, 2023-01-12:
@manuelbl , could you open the PlatformIO IDE terminal, `cd` to the project directory,  type `pio pkg update`, then restart VSCode or rebuild the Intellisense index? Does it work now?

manuelbl, 2023-01-12:
Thanks, that has fixed it. Why is the manual update needed?

ivankravets, 2023-01-13:
Thanks, fixed in https://github.com/platformio/platform-ststm32/commit/f5408ad97642b8371b0021321ac78d11df85ff90

The manual update was need for the case when previous version of package was already installed.