platformio/platformio-vscode-ide #3374: Start PIO Home Server

Allan-py, 2022-11-22:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.73.1
  PIO IDE: v2.5.5
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: getaddrinfo ENOTFOUND asynctcp
	at c:\Users\amoos\.vscode\extensions\platformio.platformio-ide-2.5.5-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10078
	at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
  ```

ivankravets, 2022-12-26:
Please re-test with the pre-release version. See the instructions below:

[image]

Does it work for you?

closed 2022-12-26 by ivankravets

Allan-py, 2022-12-31:
It seems to work. Thank You.