platformio/platformio-vscode-ide #3380: Request for a new feature (not Issues)

Bighoneypot, 2022-11-25:
@ivankravets 

In the next release if possible to add button to upload only sketch without rebuild it?
In Platformio pio to upload without rebuild command is run -t nobuild -t upload, but a new button is very simple and very Easy.

I propose the lightning bolt icon as a button , a button visible if and only if the .bin file is present in the build folder

Similar
[image]

ivankravets, 2022-11-25:
What is the reason for this button? PlatformIO uses incremental build and does not build projects every time on each upload command.

Bighoneypot, 2022-11-25:
Hi, thanks for your reply.  The sense is to flash the firmware.bin in a
very very fast way, almost like the expressif tool which is very fast.  I
practically indicated with a lightning bolt on the new button to indicate a
lightning-fast loading of the firmware.  I have been using platformio for
many years and I must say that I miss this lightning-fast loading function
and I would love it.  Pressing the lightning is having a firmware.bin
upload is a dream.

ivankravets, 2022-11-25:
Please vote for https://github.com/platformio/platformio-vscode-ide/issues/1697

closed 2022-11-25 by ivankravets