platformio/platformio-vscode-ide #3387: PIO Home not start at all

Blakinas, 2022-12-02:
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

PS C:\Users\tomas> pio home
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try  
again.
At line:1 char:1
+ pio home
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
PS C:\Users\tomas> pio home
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try 
again.
At line:1 char:1
+ pio home
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
PS C:\Users\tomas>

ivankravets, 2022-12-26:
Please re-test with the pre-release version. See the instructions below:

[image]

Does it work for you?

closed 2022-12-26 by ivankravets

steve6375, 2023-06-12:
In my case, my AV s/w (Comodo) blocked the pio.exe file so it was not copied to the folder at all.
So check for pio.exe on your hard disk and if not there, suspect your AV has blocked it.