platformio/platformio-vscode-ide #3399: Clicking build or upload buttons does nothing

cdthomp1, 2022-12-23:
Hello, 
I am working on a project and trying to upload to my Teensy 4.0 board. I was able to upload just for a few uploads, but now when I try to build or upload the code nothing happens. The terminal just prints: 
```
 *  Executing task: platformio run --target upload 

 *  Terminal will be reused by tasks, press any key to close it. 
```

Here is my INI file: 
```
[env:teensy40]
platform = teensy
board = teensy40
framework = arduino
build_flags = -D TEENSY_OPT_SMALLEST_CODE
upload_protocol = teensy-gui
lib_deps = 
	adafruit/Adafruit Unified Sensor@^1.1.7
	adafruit/Adafruit BNO055@^1.6.1
	adafruit/Adafruit BusIO@^1.14.1
	Wire
	SPI
port = /dev/cu.usbmodem124333201
```

Let me know if there is anything else I can screen grab or copy to help diagnose this issue.

Thanks! 
[image]

cdthomp1, 2022-12-23:
Restarting and checking for updates did allow the upload button to work again, but now I am getting an unknown upload error, `*** [upload] Error -1`

Here is the whole output: 

```
*  Executing task: platformio run --target upload 

Warning! Ignore unknown configuration option `port` in section [env:teensy40]
Processing teensy40 (platform: teensy; board: teensy40; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.17.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 99 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit Unified Sensor @ 1.1.7
|-- Adafruit BNO055 @ 1.6.1
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 1.0
|   |   |-- SPI @ 1.0
|   |-- Wire @ 1.0
|   |-- Adafruit Unified Sensor @ 1.1.7
|-- Adafruit BusIO @ 1.14.1
|   |-- Wire @ 1.0
|   |-- SPI @ 1.0
|-- Wire @ 1.0
|-- SPI @ 1.0
|-- BNO055
|   |-- Adafruit BNO055 @ 1.6.1
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- Wire @ 1.0
|   |   |-- Adafruit Unified Sensor @ 1.1.7
|   |-- SPI @ 1.0
|   |-- Wire @ 1.0
|-- RBGLED
Building in release mode
Compiling .pio/build/teensy40/src/main.cpp.o
Compiling .pio/build/teensy40/lib045/Adafruit Unified Sensor/Adafruit_Sensor.cpp.o
Compiling .pio/build/teensy40/lib7fa/Wire/Wire.cpp.o
Compiling .pio/build/teensy40/lib7fa/Wire/WireIMXRT.cpp.o
Compiling .pio/build/teensy40/lib7fa/Wire/WireKinetis.cpp.o
Archiving .pio/build/teensy40/lib045/libAdafruit Unified Sensor.a
Indexing .pio/build/teensy40/lib045/libAdafruit Unified Sensor.a
Compiling .pio/build/teensy40/lib7fa/Wire/utility/twi.c.o
Compiling .pio/build/teensy40/libb2a/SPI/SPI.cpp.o
Compiling .pio/build/teensy40/lib542/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/teensy40/lib542/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
Archiving .pio/build/teensy40/lib7fa/libWire.a
Indexing .pio/build/teensy40/lib7fa/libWire.a
Compiling .pio/build/teensy40/lib542/Adafruit BusIO/Adafruit_SPIDevice.cpp.o
Compiling .pio/build/teensy40/lib01c/Adafruit BNO055/Adafruit_BNO055.cpp.o
Compiling .pio/build/teensy40/libc53/BNO055/BNO055.cpp.o
Archiving .pio/build/teensy40/lib542/libAdafruit BusIO.a
Indexing .pio/build/teensy40/lib542/libAdafruit BusIO.a
Archiving .pio/build/teensy40/libb2a/libSPI.a
Indexing .pio/build/teensy40/libb2a/libSPI.a
Compiling .pio/build/teensy40/lib59b/RBGLED/REGBLED.cpp.o
Compiling .pio/build/teensy40/FrameworkArduino/AudioStream.cpp.o
Archiving .pio/build/teensy40/libc53/libBNO055.a
[... 76 lines cut ...]
Compiling .pio/build/teensy40/FrameworkArduino/usb_midi.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_mouse.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_mtp.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_rawhid.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_seremu.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_serial.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_serial2.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_serial3.c.o
Compiling .pio/build/teensy40/FrameworkArduino/usb_touch.c.o
Compiling .pio/build/teensy40/FrameworkArduino/yield.cpp.o
Archiving .pio/build/teensy40/libFrameworkArduino.a
Indexing .pio/build/teensy40/libFrameworkArduino.a
Linking .pio/build/teensy40/firmware.elf
Checking size .pio/build/teensy40/firmware.elf
Building .pio/build/teensy40/firmware.hex
Configuring upload protocol...
AVAILABLE: jlink, teensy-cli, teensy-gui
CURRENT: upload_protocol = teensy-gui
Uploading .pio/build/teensy40/firmware.hex
*** [upload] Error -1
========================================================= [FAILED] Took 12.38 seconds =========================================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

ivankravets, 2022-12-26:
Please forward to https://community.platformio.org/

closed 2022-12-26 by ivankravets