platformio/platformio-vscode-ide #3408: Start PIO Home Server

Baerschen1, 2022-12-29:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.2
  PIO IDE: v2.99.3
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\Michael Schiefer\.vscode\extensions\platformio.platformio-ide-2.99.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10339)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-01-15:
Could you reproduce this issue after the VSCode restart?

ut4uss, 2023-04-18:
Error: Could not start PIO Home server: Timeout error at Timeout._onTimeout (/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:10444) at listOnTimeout (node:internal/timers:559:17) at process.processTimers (node:internal/timers:502:7)...

I tried all the methods listed above, but it didn't work. What did I do wrong? In the browser, the start page (http://127.0.0.1:8008/) works...

ivankravets, 2023-04-18:
@ut4uss, could you try to set [platformio-ide.pioHomeServerHttpPort](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-piohomeserverhttpport) to `45001`? Also, does http://127.0.0.1:45001/ work if you start IDE?

ut4uss, 2023-04-18:
It does not work... In the "settings.yson" file, I set the port 8008, 45001, 45002 ...

 "platformio-ide.pioHomeServerHttpPort": 45001,
 "platformio-ide.pioHomeServerHttpHost": "127.0.0.1",

ut4uss, 2023-04-22:
Шалом Шаббат!
Ось на що лається налагоджувальна консоль VScode, інструмент аналізу: Debugging
information is available via VSCode > Help > Toggle Developer Tools >
Console.
Failed to install PlatformIO IDE.

workbench.desktop.main.js:90 [Extension Host] Error: PlatformIO Core has
not been installed yet!
at u.value
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:802556)
at async j.check
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/dist/extension.js:1:10439)
at async
/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/dist/extension.js:1:45062
T @ workbench.desktop.main.js:90
workbench.desktop.main.js:620   ERR [Extension Host] (node:2201) [DEP0005]
DeprecationWarning: Buffer() is deprecated due to security and usability
issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or
Buffer.from() methods instead.
(Use `exe --trace-deprecation ...` to show where the warning was created)
workbench.desktop.main.js:90 [Extension Host] (node:2201) [DEP0005]
DeprecationWarning: Buffer() is deprecated due to security and usability
issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or
Buffer.from() methods instead.
(Use `exe --trace-deprecation ...` to show where the warning was created)
T @ workbench.desktop.main.js:90
workbench.desktop.main.js:620   ERR [Extension Host] Error: bad_status
at
/home/duke/.vscode/extensions/kiteco.kite-0.147.0/dist/kite-extension.js:2:53954
at process.processTicksAndRejections
(node:internal/process/task_queues:96:5)
workbench.desktop.main.js:90 [Extension Host] Error: bad_status
at
/home/duke/.vscode/extensions/kiteco.kite-0.147.0/dist/kite-extension.js:2:53954
at process.processTicksAndRejections
(node:internal/process/task_queues:96:5)
T @ workbench.desktop.main.js:90
workbench.desktop.main.js:1904 Overwriting grammar scope name to file
mapping for scope source.yaml.
Old grammar file:
file:///usr/share/code/resources/app/extensions/yaml/syntaxes/yaml.tmLanguage.json.
New grammar file:
file:///home/duke/.vscode/extensions/redhat.vscode-yaml-1.12.2/syntaxes/yaml.tmLanguage.json
register @ workbench.desktop.main.js:1904
workbench.desktop.main.js:90 [Extension Host] Error: Could not find
portable Python for linux_x86_64
at
/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:805094
at process.processTicksAndRejections
(node:internal/process/task_queues:96:5)
at async u.value
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:804487)
at async j.install
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/dist/extension.js:1:10606)
at async
/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/dist/extension.js:1:45694
T @ workbench.desktop.main.js:90
workbench.desktop.main.js:620   ERR [Extension Host] (node:2201) [DEP0147]
DeprecationWarning: In future versions of Node.js, fs.rmdir(path, {
recursive: true }) will be removed. Use fs.rm(path, { recursive: true })
instead
workbench.desktop.main.js:90 [Extension Host] (node:2201) [DEP0147]
DeprecationWarning: In future versions of Node.js, fs.rmdir(path, {
recursive: true }) will be removed. Use fs.rm(path, { recursive: true })
instead
T @ workbench.desktop.main.js:90
workbench.desktop.main.js:90 [Extension Host] runCommand /usr/bin/python3
Array(3) Object
workbench.desktop.main.js:90 [Extension Host] /usr/bin/python3 Error:
Error: The Python 3.9.2 (/usr/bin/python3) interpreter is not compatible.
[... 181 lines cut ...]
workbench.desktop.main.js:1997 [[object Object]]PlatformIO Core is not
installed
$onExtensionRuntimeError @ workbench.desktop.main.js:1997
workbench.desktop.main.js:1997 Error: PlatformIO Core is not installed
at O
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:6713)
at L
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:7114)
at BA.host
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:10522)
at new Promise (<anonymous>)
at BA
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:10397)
at process.processTicksAndRejections
(node:internal/process/task_queues:96:5)
at async Module.dA
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:10112)
at async E.getWebviewContent
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/dist/extension.js:1:4567)
at async E.newPanel
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/dist/extension.js:1:4035)
at async E.toggle
(/home/duke/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/dist/extension.js:1:3594)
at async m.h
(/usr/share/code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:98:125789)

thermseekr, 2023-05-09:
Hi Ivan, hi all

I spent most of today trying to install and run PIO. And the most weird, it's on a machine that has had it running before. First, Core refused to install, disabling the AV solved that part. Now Home installs but won't start. It times out. PIO CLI fails in the same way, and trying directly in the browser returns a page not found error. It may still be something related to the environment, as central adm keeps restricting things more and more everyday, or maybe it's something else in my PC, but I couldn't pinpoint  the cause yet.

Can you help me move forward?

Thank you in advance.

[image]

[image]

```
[Extension Host] Found compatible PlatformIO Core 6.1.7 -> C:\Users\tales.maschio\.platformio\penv\Scripts\platformio.exe
console.ts:137 [Extension Host] PIO Core State 
{core_version: '6.1.7', python_version: '3.11.3', core_dir: 'C:\\Users\\tales.maschio\\.platformio', cache_dir: 'C:\\Users\\tales.maschio\\.platformio\\.cache', penv_dir: 'C:\\Users\\tales.maschio\\.platformio\\penv', …}
console.ts:137 [Extension Host] runCommand C:\Users\tales.maschio\.platformio\penv\Scripts\python.exe 
(14) ['-m', 'platformio', '-c', 'vscode', 'home', '--port', 8008, '--host', '127.0.0.1', '--session-id', '23da7162f70a651f99bed7b90e58810156493ed9', '--shutdown-timeout', 3600, '--no-open']
 
{_id: 'C:\\Users\\tales.maschio\\.platformio\\penv\\Scripts\\python.exe-0.5862289237306766'}
console.ts:137 [Extension Host] Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\tales.maschio\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10444)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
console.ts:137 [Extension Host] runCommand C:\Users\tales.maschio\.platformio\penv\Scripts\python.exe 
(14) ['-m', 'platformio', '-c', 'vscode', 'home', '--port', 8008, '--host', '127.0.0.1', '--session-id', '23da7162f70a651f99bed7b90e58810156493ed9', '--shutdown-timeout', 3600, '--no-open']
 
{_id: 'C:\\Users\\tales.maschio\\.platformio\\penv\\Scripts\\python.exe-0.11763642717472722'}
console.ts:137 [Extension Host] Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\tales.maschio\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10444)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
console.ts:137 [Extension Host] runCommand C:\Users\tales.maschio\.platformio\penv\Scripts\python.exe 
(14) ['-m', 'platformio', '-c', 'vscode', 'home', '--port', 8008, '--host', '127.0.0.1', '--session-id', '23da7162f70a651f99bed7b90e58810156493ed9', '--shutdown-timeout', 3600, '--no-open']
 
{_id: 'C:\\Users\\tales.maschio\\.platformio\\penv\\Scripts\\python.exe-0.760075246264907'}
console.ts:137 [Extension Host] Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\tales.maschio\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10444)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
notificationsAlerts.ts:42 Error: Could not start PIO Home server: Timeout error 	at Timeout._onTimeout (c:\Users\tales.maschio\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10444) 	at listOnTimeout (node:internal/timers:559:17) 	at process.processTimers (node:internal/timers:502:7)...
log.ts:441   ERR [Extension Host] Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\tales.maschio\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10444)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
console.ts:137 [Extension Host] Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\tales.maschio\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10444)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
```

ivankravets, 2023-05-10:
Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) in VSCode and type 
```
pio home
```

Do you see any errors?

thermseekr, 2023-05-10:
We use Bitdefender from Romania and Crowdstrike from USA. I disabled Bitdefender for testing purposes but results were the same. For disabling Crowdstrike I must ask to the central administration, haven't done that yet, but can do if needed. Our computers are part of a Windows Active Directory system but I'm local admin, maybe I should start VSCode as admin? Will try that and report back.

thermseekr, 2023-05-10:
> PlatformIO Core can't download a package/write to disk. Something blocks it. Do you use any espionage software such as russian Kaspersky?

Not that we know, at least... Other thing I can do is monitor what's happening with Wireshark. Will try that too.

ivankravets, 2023-05-10:
Please contact me@ikravets.com directly to keep this thread clean. If we find a solution, we will share it here. Thanks!

thermseekr, 2023-05-11:
Hello Ivan.

I finally managed to run PlatformIO. 

I had tried before to install using an external network connection, one that has direct access to the internet without any firewall, but at that time I had the antivirus active so PIO Core would not install. After I disabled the AV, PIO Core installed even under the protected network, but PIO Home was still not going. Today I connected back to the open network and PIO Home finished installed and ran normally.

So the conclusion is:
- PIO Core installation was being blocked by the AV, as with the AV disabled it installed even under the protected network;
- PIO Home installation was being blocked by the corporate FW, as even with the AV disabled it would not install under the protected network, but installed and ran normally under the external network even with the AV active.

After PIO Home finished its thing I returned to the protected network and all seems to be running well. I will report these problems to the people running our AV and FW systems and ask them to check what's wrong.

Thank you very much for your support. Best regards from Brazil.

Tales

ivankravets, 2023-05-11:
@thermseekr, do you have access to the following resources when FW is enabled?
- https://platformio.org
- https://registry.platformio.org

thermseekr, 2023-05-12:
@ivankravets yes, positive for both.

danipaez09, 2023-05-15:
Hi, i have this error

[image]

I am trying to create a project for an STM card in an mbed environment. I created the folder by executing the 'pio home' command, however when I open it in VSCode, the folders that relate to the mbed do not appear in the 'src' folder. try to reinstall the VSCode and python and nothing

Kymaro, 2023-05-23:
> @thermseekr, do you have access to the following resources when FW is enabled?
> 
> * https://platformio.org
> * https://registry.platformio.org

Hi, I have the same issue here. I can access to those website but can't use PIO in VSCode. From what I understand, VSCode doesn't use the  custom Windows SSL Certificate my company installed.

ivankravets, 2023-05-23:
Have you disabled "Http: Proxy Strict SSL"? See https://docs.platformio.org/en/latest/integration/ide/vscode.html#proxy-server-support

Please disable and restart VSCode.

Kymaro, 2023-05-24:
Works fine, thanks for the info.

humid-sirocco, 2023-06-15:
Hi friends. If you find yourself experiencing issues with corporate CA certificate not being recognized, I'd recommend use of the win-ca VScode package, its sole purpose is exposing the windows CA certificates to vscode extensions. No need to disable verification of the proxy strict SSL.

iottrends, 2023-07-05:
Any solution for this??
I am doing it on ubuntu 20.4
root@labuser-OptiPlex-5050:/home/labuser/Downloads# python3 get-platformio.py
Installer version: 1.1.3
Platform: Linux-5.0.0-23-generic-x86_64-with-glibc2.25
Python version: 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /root/.platformio/penv
/usr/bin/python3
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues

ivankravets, 2023-07-05:
@iottrends why do you run PlatformIO from the `root` user? It is insecure. Please check related issue https://github.com/platformio/platformio-core-installer/issues/85

ivankravets, 2023-07-05:
Could you remove `~/.platformio/penv` folder and restart VSCode? It should be fixed now.

niboon39, 2024-01-10:
Hi

I can run it. 
I installed python 3.6.15 and python get_platformio.py.

# Python 3.6.15 
https://stackoverflow.com/questions/72102435/how-to-install-python3-6-on-ubuntu-22-04

# Python get_platformio.py
https://raw.githubusercontent.co

[image]

niboon39, 2024-01-11:
> Any solution for this?? I am doing it on ubuntu 20.4 root@labuser-OptiPlex-5050:/home/labuser/Downloads# python3 get-platformio.py Installer version: 1.1.3 Platform: [...]

Try this

ivankravets, 2024-01-12:
@niboon39 could your issue be linked with https://github.com/platformio/platformio-core-installer/issues/85 ?

djswirl, 2024-01-12:
I had the smae problem, thanks to AI all sorted, these are the steps I took

CMD: pio update

CMD: pip uninstall platformio
CMD: pip install -U platformio

Open VScode and install PIO Extension again

JosefWN, 2024-01-14:
Is anyone looking into this? The ticket is a year old but it happened to me today, I just installed PlatformIO for the first time. If it's unstable I'm hesitant to spend time on onboarding myself.

ivankravets, 2024-01-17:
@JosefWN, have you resoled your issue? What is your OS?

JosefWN, 2024-01-17:
MacOS, but I already removed it, maybe I will give it a spin in the future!

ivankravets, 2024-01-18:
We updated built-in Python for macOS. It should solve your issue. Please remove `~/.platformio` folder and install PlatformIO IDE Extension.

JosefWN, 2024-01-18:
Where did you do that? The latest commit is from last month in this repo, and the latest release for the VS Code plugin was 2023-12-23? I installed it for the first time 4 days ago. Anyhow, I am not spending more time on this for now.

OmarBannour, 2024-07-12:
Error: Could not start PIO Home server: Timeout error
  at Timeout._onTimeout (c:\Users\Igor\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:182:9)
  at listOnTimeout (node:internal/timers:559:17)
  at processTimers (node:internal/timers:502:7)   i need help

Buom01, 2024-07-26:
Removed `platformio-core` package solved the issue on my CachyOS (Arch based) system.

HazemRagabElsayed, 2024-07-29:
[image]

HazemRagabElsayed, 2024-07-29:
I have  a problem with this platformio

HazemRagabElsayed, 2024-07-29:
[image]

HazemRagabElsayed, 2024-07-30:
> @HazemRagabElsayed <https://github.com/HazemRagabElsayed> , do you use
> espionage software (some call it "antivirus")? Please disable it for a
> while or add PlatformIO to the whitelist. It seems it blocks PlatformIO's
> subprocess PIO Home.

Yes I disabled all of defenders(security and firewall defender) and same
problem is encountered

HazemRagabElsayed, 2024-08-06:
> @HazemRagabElsayed , do you use espionage software (some call it "antivirus")? Please disable it for a while or add PlatformIO to the whitelist. It seems it blocks PlatformIO's subprocess PIO Home.

Problem is solved it was because I installed Atmel studio and when I delete it and try to install PlatformIO again it is installed successfully

LoanHanu, 2024-09-12:
I got the same warnings.
Is there no any solution to set up PlatformIO environment in offline computer?

shubham-vunet, 2024-09-21:
For me the issue was fixed by removing an environment variable `SSLKEYLOGFILE` from the Windows.

Patuteee, 2024-11-29:
Al escribir "pio home" me aparece lo siguiente en rojo:

pio : El término 'pio' no se reconoce como nombre de un cmdlet, función, archivo de script o 
programa ejecutable. Compruebe si escribió correctamente el nombre o, si incluyó una ruta de   
acceso, compruebe que dicha ruta es correcta e inténtelo de nuevo.
En línea: 1 Carácter: 1
+ pio home
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException        
    + FullyQualifiedErrorId : CommandNotFoundException

yijun223, 2024-12-31:
[image]

arduinka55055, 2025-01-15:
this helped me
>n the file C:\Users "Username"\AppData\Roaming\Code\User\settings.json needs to be checked "platformio-ide.pioHomeServerHttpPort": 8008

https://github.com/platformio/platformio-vscode-ide/issues/3954

Rosemajor, 2025-03-15:
还有一种解决办法无需重启电脑即可解决：python源的问题，首先挂全局梯子（好用一点二的），然后将源切换回默认源
pip config set global.index-url https://pypi.org/simple
，重启vscode即可安装。

KarlHabsburg, 2025-03-18:
We just had the same Issue in our Corporate Environment.

If you're using a Proxy which does SSL-Interception you have to add the Root Certificate from the Proxy, for example ZScaler, to the following File: "C:\Users\%username%\.platformio\penv\Lib\site-packages\certifi\cacert.pem"

ivankravets, 2025-03-21:
@KarlHabsburg , could you try `REQUESTS_CA_BUNDLE`? See https://requests.readthedocs.io/en/latest/user/advanced/#ssl-cert-verification

tobiaswaelde, 2025-04-15:
I completely uninstalled and reinstalled PIO.
I still get this error message:

Error: Could not start PIO Home server: Timeout error
at Timeout._onTimeout (c:\Users\tobiaswaelde.vscode\extensions\platformio.platformio-ide-3.3.4-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10741)
at listOnTimeout (node:internal/timers:581:17)
at processTimers (node:internal/timers:519:7)...

embedded4ever, 2025-05-01:
I have faced same error too, and this solution is worked for me, 

- If you use a proxy or custom SSL certificate, please disable [Http: Proxy Strict SSL](https://docs.platformio.org/en/latest/integration/ide/vscode.html#proxy-server-support).

brylleee, 2025-05-12:
`pio home` works and opens PlatformIO Home on my default browser, but the same error pops up at VSCode and the PlatformIO IDE does not open.

OldMicroGuy, 2025-05-19:
I'm getting this error also.  Worked fine on another computer, but I had to upgrade to a new laptop (Windows 11 Home).
Currently, every firewall and protection setting is turned off, trying to get it to share with other devices on my personal network.  (I despise Windows for this). Security will be turned back on once I copy a bunch of stuff.
The file the error complains about in the timeout, exists, but that's all I can say about it.
I can install VS OK, and can manually select the PIO from the extensions list.
During the install, I also get an error (every time) that the PIO installer has been halted because it's running in another window, but I can't seem to find it running anywhere else.
I also have no command line interface.  Any commands such as "pio home", "pio anything" fail with "not a valid command".

VS also installed with Android Studio also installed OK.  So VS is up and running.  I did look at "known issues" but didn't see anything that helped.  (and yep, it's been uninstalled and re-installed multiple times).  Strict SSL has also been unchecked, with no results.

ayavilevich, 2025-05-19:
> I also have no command line interface.

Try:
PLATFORMIO -> QUICK ACCESS -> Miscellaneous -> PlatformIO Core CLI

ayavilevich, 2025-05-19:
Normally I am using the PIO extension and it works just fine. Suddenly home stopped working from VSCode. "pio home" worked.
Since then I have restarted the machine and it is all back to normal now. So if you are having this, try a restart.

OldMicroGuy, 2025-05-19:
> VS also installed with Android Studio also installed OK. So VS is up and running. I did look at "known issues" but didn't see anything that helped. (and yep, it's been uninstalled and re-installed multiple times). Strict SSL has also been unchecked, with no results.

Success. I just deleted all old references to anything PlatformIO and re-installed it.  The real error was the "can't install", not the timeout.

RanchoHam, 2025-07-25:
I had this timeout occur again after a VS Code update.  I closed and restarted VS Code, but still the timeout occurred.  I rebooted the machine, and now it works normally.  CLI `pio home` worked to open a Home in a normal browser.  

I have been having problems with browsers too aggressively caching data.  This could be related to that, but for now may I suggest adding an item 5 to the suggestion list:  "5. Reboot the machine."

hugoo0, 2025-11-24:
> # Solutions
> 
>     1. Try to restart VSCode. Very often it helps.
> 
>     2. If you use a proxy or custom SSL certificate, please disable [Http: Proxy Strict SSL](https://docs.platformio.org/en/latest/integration/ide/vscode.html#proxy-server-support).
> 
>     3. If you use custom antivirus tools or firewalls, please ensure that they don't block PlatformIO. You will need to add [PlatformIO Core Directory](https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/core_dir.html) to the "whitelist".
> 
>     4. Open [PlatformIO CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli), and type `pio home`. The PIO Home should be opened in your default web browser (which means all packages are downloaded). Restart VSCode.
> 
> 
> If the steps above didn't help you, please leave a comment below with further details and we will help you ASAP! 🤝
> ### Exception
> 
> ```
> Error: Could not start PIO Home server: Timeout error
>   at Timeout._onTimeout (c:\Users\Michael Schiefer\.vscode\extensions\platformio.platformio-ide-2.99.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10339)
>   at listOnTimeout (node:internal/timers:559:17)
>   at process.processTimers (node:internal/timers:502:7)
> ```

mount33, 2026-11-02:
If your WLS is running, try change default PIO Home Server Http Port to 8008 in VSCode setting.