platformio/platformio-vscode-ide #3415: Start PIO Home Server

aryan-ir, 2023-01-01:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.2
  PIO IDE: v2.5.5
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect ETIMEDOUT 127.0.0.1:8010
	at c:\Users\A.Nazari\.vscode\extensions\platformio.platformio-ide-2.5.5-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10078
	at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
  ```

ivankravets, 2023-01-15:
Could you switch to the [PlatformIO IDE 3.0 Preview](https://community.platformio.org/t/platformio-ide-3-0-preview-for-vscode-we-want-your-feedback/31151)? Does PIO Home work now?

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets