platformio/platformio-vscode-ide #3417: Start PIO Home Server

JosephAbouElOula, 2023-01-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.2
  PIO IDE: v2.5.5
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: ERROR:    [Errno 10048] error while attempting to bind on address ('127.0.0.1', 8010): only one usage of each socket address (protocol/network address/port) is normally permitted

	at c:\Users\joseph.aboueloula\.vscode\extensions\platformio.platformio-ide-2.5.5-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:9996
	at ChildProcess.o (c:\Users\joseph.aboueloula\.vscode\extensions\platformio.platformio-ide-2.5.5-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4691)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-01-15:
Could you switch to the [PlatformIO IDE 3.0 Preview](https://community.platformio.org/t/platformio-ide-3-0-preview-for-vscode-we-want-your-feedback/31151)? Does PIO Home work now?

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets