platformio/platformio-vscode-ide #3431: How to disable install PlatformIO IDE automatically?

ysong-aceinna, 2023-01-16:
Hi, I want to disable automatically install PlatformIO IDE, it always failed. My PlatformIO IDE works well so far and I don't want to update it to new version, thanks.

[image]

makefu, 2023-01-28:
Hey, i am struggling with the same issue, i have a working platformio instance installed and i hoped the plugin would just use the executable if available.

ivankravets, 2023-02-01:
The new PlatformIO IDE 3.0 is out https://github.com/platformio/platformio-vscode-ide/releases/tag/v3.0.0

If you don't want to use built-in PlatformIO Core, please set [platformio-ide.useBuiltinPIOCore](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usebuiltinpiocore) to `false`.

P.S.: We don't recommend disabling this option.

closed 2023-02-01 by ivankravets

makefu, 2023-02-03:
Hi @ivankravets thanks for the response, unfortunately updating to 3.0.0 didn't solve the issue for me. i have a working platformio installation in my PATH, i tried to set useBuiltinPIOCore to false and setting the custom PATH, however the issue still occurs.

ivankravets, 2023-02-03:
@makefu, I use similar configuration everyday (custom PATH + disabled built-in PIO Core). Could you share here your VSCode's `settings.json`?

makefu, 2023-02-03:
```
~/r/bureautomation/rgbstripe which pio
/etc/profiles/per-user/makefu/bin/pio        # a symlink to /nix/store/br46kmhdnqmxjnrmaigc2hgh7gz7jbwm-platformio/bin/platformio

```~/r/bureautomation/rgbstripe echo $PATH
/nix/store/7z7blr44ix77fxck3ihgyn6mh92ms9cm-glib-2.74.3-bin/bin:/run/wrappers/bin:/home/makefu/.nix-profile/bin:/etc/profiles/per-user/makefu/bin:/nix/var/nix/profiles/default/bin:/run/current-system/sw/bin
```
settings.json:
```
{
    "workbench.editorAssociations": {
        "*.ipynb": "jupyter-notebook"
    },
    "notebook.cellToolbarLocation": {
        "default": "right",
        "jupyter-notebook": "left"
    },
    "window.zoomLevel": -1,
    "platformio-ide.useBuiltinPIOCore": false,
    "platformio-ide.useBuiltinPython": false,
    "platformio-ide.useDevelopmentPIOCore": false,
    "platformio-ide.customPATH": "/nix/store/6g7rswavw1qk4h3rjhdh6fdrmlpn9cjw-platformio/bin/platformio", # also tried /nix/store/6g7rswavw1qk4h3rjhdh6fdrmlpn9cjw-platformio/bin and /etc/profiles/per-user/makefu/bin/
    "platformio-ide.pioHomeServerHttpPort": 45000,
    "platformio-ide.disablePIOHomeStartup": true
}
```

how is the pio installation detection performed? it looks like it is more complicated than `command -v pio` for PATH lookup

ivankravets, 2023-02-03:
Please read carefully https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-custompath

It MUST be the path to the FOLDER where `pio/pio.exe` is installed. BUT NOT A PATH TO THE PIO.EXE.

So, please update `platformio-ide.customPATH` to `/nix/store/br46kmhdnqmxjnrmaigc2hgh7gz7jbwm-platformio/bin/`, restart  VSCode and open Help > Toggle Developer Tools > Console. You will see all progress. Please share it here.

makefu, 2023-02-04:
Hi! I think i've found the culprit. as i wrote above i tried setting the path to `bin/` before, however the developer console revealed that for checking if platformio is available some code is running `import platformio`, however the python interpreter i use has no access to the platformio python library which in turn results in the error message.
I will try to install platformio library for python and come back

ivankravets, 2023-02-05:
@makefu, it seems you have broken the virtual Python environment. If you do everything correctly, you do not need to apply any extra steps and workarounds. Please remove [Multiple PlatformIO Cores in a system](https://docs.platformio.org/en/latest/core/installation/troubleshooting.html#multiple-platformio-cores-in-a-system) and enable built-in PlatformIO Core.