platformio/platformio-vscode-ide #3438: Start PIO Home Server

fhehgcngy, 2023-01-17:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.0
  PIO IDE: v2.99.4
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.99.4-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10392)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```

Tandough, 2023-01-20:
Uninstall platformio from VSCode, then navigate to delete the platformio dot folder (.platformio) in C:\Users\user and show hidden items in file explorer if you can't see it. Open VSCode again and reinstall platformio and it should be working.

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets