platformio/platformio-vscode-ide #3439: Start PIO Home Server

GuyzMaker, 2023-01-17:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.3
  PIO IDE: v2.5.5
  System: Linux, 5.15.0-58-generic, x64

  %23 Exception
  ```
  Error: Error: Could not load broken JSON: /home/guyz/.platformio/packages/tool-scons/.piopm

	at /home/guyz/.vscode/extensions/platformio.platformio-ide-2.5.5-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:9996
	at ChildProcess.o (/home/guyz/.vscode/extensions/platformio.platformio-ide-2.5.5-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4691)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

Tandough, 2023-01-20:
Uninstall platformio from VSCode, then navigate to delete the platformio dot folder (.platformio) in C:\Users\user and show hidden items in file explorer if you can't see it. Open VSCode again and reinstall platformio and it should be working.

GuyzMaker, 2023-01-27:
It's okay, I've got it covered.
But I'm not under windows.

closed 2023-01-27 by GuyzMaker