platformio/platformio-vscode-ide #3441: Start PIO Home Server

ujakkal, 2023-01-19:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.3
  PIO IDE: v2.99.4
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\Uday.Jakkal\.vscode\extensions\platformio.platformio-ide-2.99.4-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10392)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```

Tandough, 2023-01-20:
Uninstall platformio from VSCode, then navigate to delete the platformio dot folder (.platformio) in C:\Users\user and show hidden items in file explorer if you can't see it. Open VSCode again and reinstall platformio and it should be working.

ivankravets, 2023-02-01:
Do you use any antivirus tools?

ujakkal, 2023-02-06:
> CAUTION: This email originated from outside of Vertiv. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi Ivan,

By system is protected with Cybereason along side WindowsDefender deployed by my organization.

Please let me know if there is any quick-fix to bypass the same.

Best regards,
Uday Jakkal

CONFIDENTIALITY NOTICE: This e-mail and any files transmitted with it are intended solely for the use of the individual or entity to whom they are addressed and may contain confidential and privileged information protected by law. If you received this e-mail in error, any review, use, dissemination, distribution, or copying of the e-mail is strictly prohibited. Please notify the sender immediately by return e-mail and delete all copies from your system.

ivankravets, 2023-02-06:
Is it possible to add `C:/Users/%YOURNAME%/.platformio/penv` folder to the white list?

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets