platformio/platformio-vscode-ide #3443: Debugging ESP32 project with multiple threads can be very slow

drorgl, 2023-01-20:
For example a project with 20 threads, totaling 26 threads (main, IDLE, IDLE, ipc0, ipc1, esp_timer) can take 6-8 seconds to step.

I've analyzed the problem and it seems its in this package or in `platformio-vscode-debug`, while the source code was not available it does seem similar to [Enneary/module-debug](https://github.com/Enneary/module-debug).

Digging in the requests that cause the slow down in GDB, it seems that even when the `REGISTERS` and `CALL STACK` debugger panes are collapsed, the plugin still requests the underlying `register-values` and `threads`. 

The `threads` or `info threads` is very slow in ESP32/GDB and takes most of the time between `step` / `next`.

It would seem like an easy fix to speed things up for someone who understands how VSCode debugger extension works: check if the pane is open and if not, don't request the data needed to update it.

Attempting to increase the `debug_speed` didn't help.

While you're at it, add a check for `PERIPHERALS` pane as well.

ivankravets, 2023-02-01:
Could you try to collapse "Registers" + "Peripheral" views and restart VScode? It should work.

P.S: We plan soon to rewrite from scratch debugging component with a lot of new features and improvements. Sorry for the current issues.

james-h-dt, 2023-08-15:
Collapsing "Registers" and "Peripheral" view + restar VSC did not improve debug speed for me. 
Also tried collapsing all views in the debug panel (except the essentials: Breakpoints and Watch). Same behaviour seen.
I was using a very simple single-threaded test app. 
Each debug step took 3 - 5 seconds.

YutingYou, 2023-10-18:
Same for me, using esp32s3 build-in jtag

vscode + espidf debug: step debug is very slow, 1~2 seconds
vscode + cppdbg: normal speed
terminal + openocd +esp-gdb: normal speed

Any thought?

SnazzyBlanket, 2024-05-04:
any update on this? extremely slow for me as well.

ghisbo, 2025-03-15:
> Could you try to collapse "Registers" + "Peripheral" views and restart VScode? It should work.
> 
> P.S: We plan soon to rewrite from scratch debugging component with a lot of new features and improvements. Sorry for the current issues.

What is the status of the "rewrite" for the debugging component?