platformio/platformio-vscode-ide #3447: Start PIO Home Server

phobicdotno, 2023-01-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.3
  PIO IDE: v2.99.4
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
    at Timeout._onTimeout (c:\Users\kkd\.vscode\extensions\platformio.platformio-ide-2.99.4-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:182:9)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-02-01:
Do you use any antivirus tools?

phobicdotno, 2023-02-01:
Hey. Only Windows Defender

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets