platformio/platformio-vscode-ide #3453: Start PIO Home Server

danielvegafer, 2023-01-30:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.65.2
  PIO IDE: v2.99.4
  System: Windows_NT, 10.0.10240, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\dvega\.vscode\extensions\platformio.platformio-ide-2.99.4\node_modules\platformio-node-helpers\dist\index.js:1:10392)
	at listOnTimeout (internal/timers.js:554:17)
	at processTimers (internal/timers.js:497:7)
  ```

ivankravets, 2023-02-01:
Espionage rusish Kaspersky antivirus or something better? Could you try to disable it, restart VSCode, does PIO Home work now?

danielvegafer, 2023-02-01:
Karspersky yes. dont works with antivirus disabled

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets