platformio/platformio-vscode-ide #3457: Failure to start PIO Home server: Timeout

jcaldwe8, 2023-02-01:
Configuration

  VSCode: 1.74.3
  PIO IDE: v2.99.4
  System: Windows_NT, 10.0.22621, x64

I receive the following error when trying to install both the release and pre-release version of PlatformIO
  Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\caldw\.vscode\extensions\platformio.platformio-ide-2.99.4-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10392)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```
From the Developer Tools, I see the following errors
`PlatformIO Core is not installed`
`Compatible PlatformIO Core not found.`

Additionally, when trying to run 'pio home' from the PlatformIO CLI, I receive the following error

```
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path
was included, verify that the path is correct and try again.
At line:1 char:1
+ pio home
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

I do not have platformio in my system variables and I have tried uninstalling, deleting C:\User\<user>\.platformio, reinstalling PlatformIO and refreshing VS Code after deleting C:\User\<user>\.platformio. 
Also, I have tried a installing PlatformIO after a clean installation of VS Code.
I have also disabled Windows Antivirus during a reinstallation and still receive the errors above.

ivankravets, 2023-02-01:
Do you have any custom PlatformIO settings in the VSCode? Maybe, you disabled Built-in PlatformIO Core?  https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usebuiltinpiocore

jcaldwe8, 2023-02-01:
I've just checked and my Use a built-in PlatformIO Core setting is set to true.

ivankravets, 2023-02-02:
Could you provide a remote session under your control to debug this issue? Please contact me@ikravets.com.

jcaldwe8, 2023-02-08:
This was resolved by removing broken Python paths and uninstalling Python.

closed 2023-02-08 by jcaldwe8