platformio/platformio-vscode-ide #3473: Start PIO Home Server

durveshodi, 2023-02-04:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.75.0
  PIO IDE: v3.0.0
  System: Linux, 5.4.0-1100-azure, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
    at Timeout._onTimeout (/home/codespace/.vscode-remote/extensions/platformio.platformio-ide-3.0.0-linux-x64/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/home.js:181:22)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-02-06:
Do you use antivirus tools?

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets