platformio/platformio-vscode-ide #3475: Start PIO Home Server

zhujipeng, 2023-02-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.70.0
  PIO IDE: v3.0.0
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10392)
	at listOnTimeout (node:internal/timers:557:17)
	at processTimers (node:internal/timers:500:7)
  ```

ivankravets, 2023-02-06:
Do you use antivirus tools?

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets