platformio/platformio-vscode-ide #3477: Creating new project results in high CPU, and just hangs on setup

CropWatchDevelopment, 2023-02-05:
Hello,
I am attempting to create a new peoject, however, when I do, platformio just hangs.
I am running Linux Mint:
System:
  Kernel: 5.15.0-58-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.6.7
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy
Machine:
  Type: Detachable System: Panasonic product: CF-33S-1 v: 001 serial: <superuser required>
    Chassis: type: 32 v: 001 serial: <superuser required>
  Mobo: Panasonic model: CF33S-1 v: 1 serial: <superuser required> UEFI: American Megatrends
    v: 1.50L23 date: 07/28/2020
Battery:
  ID-1: BAT1 charge: 39.1 Wh (100.0%) condition: 39.1/44.5 Wh (88.0%) volts: 12.4 min: 10.8
    model: Panasonic CF-VZSU1B serial: <filter> status: N/A
  ID-2: BAT2 charge: 26.5 Wh (65.6%) condition: 40.4/44.5 Wh (90.9%) volts: 10.9 min: 10.8
    model: Panasonic CF-VZSU1B serial: <filter> status: Discharging
CPU:
  Info: dual core model: Intel Core i5-6300U bits: 64 type: MT MCP arch: Skylake rev: 3 cache:
    L1: 128 KiB L2: 512 KiB L3: 3 MiB
  Speed (MHz): avg: 1813 high: 1848 min/max: 400/3000 cores: 1: 1800 2: 1800 3: 1806 4: 1848
    bogomips: 19999
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel Skylake GT2 [HD Graphics 520] vendor: Matsushita driver: i915 v: kernel ports:
    active: eDP-1 empty: HDMI-A-1 bus-ID: 00:02.0 chip-ID: 8086:1916
  Device-2: Acer USB HD Webcam type: USB driver: uvcvideo bus-ID: 1-10:11 chip-ID: 5986:0689
  Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: modesetting unloaded: fbdev,vesa
    gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2880x1920 s-dpi: 144
  Monitor-1: eDP-1 res: 2880x1920 dpi: 288 diag: 305mm (12")
  OpenGL: renderer: Mesa Intel HD Graphics 520 (SKL GT2) v: 4.6 Mesa 22.2.5 direct render: Yes
Audio:
  Device-1: Intel Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Imaging Unit
    vendor: Matsushita driver: ipu3-imgu bus-ID: 00:05.0 chip-ID: 8086:1919
  Device-2: Intel CSI-2 Host vendor: Matsushita driver: ipu3-cio2 bus-ID: 00:14.3
    chip-ID: 8086:9d32
  Device-3: Intel Sunrise Point-LP HD Audio vendor: Matsushita driver: snd_hda_intel v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:9d70
  Sound Server-1: ALSA v: k5.15.0-58-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel Ethernet I219-LM vendor: Matsushita driver: e1000e v: kernel port: N/A
    bus-ID: 00:1f.6 chip-ID: 8086:156f
  IF: enp0s31f6 state: down mac: <filter>
  Device-2: Intel Wireless 8260 driver: iwlwifi v: kernel pcie: speed: 2.5 GT/s lanes: 1
    bus-ID: 01:00.0 chip-ID: 8086:24f3
  IF: wlp1s0 state: up mac: <filter>
  Device-3: Realtek RTL8153 Gigabit Ethernet Adapter type: USB driver: r8152 bus-ID: 2-3.1.1:4
    chip-ID: 0bda:8153
  IF: enxbcc342df2221 state: down mac: <filter>
  IF-ID-1: br-ca4455f4c2b0 state: down mac: <filter>
  IF-ID-2: docker0 state: down mac: <filter>
  IF-ID-3: wwan0 state: down mac: <filter>
Bluetooth:
  Device-1: Intel Bluetooth wireless interface type: USB driver: btusb v: 0.8 bus-ID: 1-8:9
    chip-ID: 8087:0a2b
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.2
    sub-v: 100
Drives:
  Local Storage: total: 238.47 GiB used: 198.53 GiB (83.2%)
  ID-1: /dev/sda vendor: Toshiba model: THNSNK256GVN8 size: 238.47 GiB speed: 6.0 Gb/s
    serial: <filter>
Partition:
  ID-1: / size: 233.18 GiB used: 198.52 GiB (85.1%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 511 MiB used: 5.2 MiB (1.0%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 63.8 MiB (3.1%) priority: -2 file: /swapfile
Sensors:
  System Temperatures: cpu: 73.6 C pch: 63.5 C mobo: 73.6 C
  Fan Speeds (RPM): N/A
Repos:
  Packages: 3450 apt: 3430 flatpak: 16 snap: 4
  No active apt repos in: /etc/apt/sources.list
  No active apt repos in: /etc/apt/sources.list.d/additional-repositories.list
  Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list
    1: deb [signed-by=/usr/share/keyrings/brave-browser-archive-keyring.gpg arch=amd64] https: //brave-browser-apt-release.s3.brave.com/ stable main
  No active apt repos in: /etc/apt/sources.list.d/docker.list
  Active apt repos in: /etc/apt/sources.list.d/github-cli.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https: //cli.github.com/packages stable main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/nodesource.list
    1: deb [signed-by=/usr/share/keyrings/nodesource.gpg] https: //deb.nodesource.com/node_16.x jammy main
    2: deb-src [signed-by=/usr/share/keyrings/nodesource.gpg] https: //deb.nodesource.com/node_16.x jammy main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com vera main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  No active apt repos in: /etc/apt/sources.list.d/pistache_team-stable-focal.list
  No active apt repos in: /etc/apt/sources.list.d/pistache_team-unstable-focal.list
  No active apt repos in: /etc/apt/sources.list.d/rethinkdb.list
  No active apt repos in: /etc/apt/sources.list.d/signal-xenial.list
  No active apt repos in: /etc/apt/sources.list.d/slack.list
  Active apt repos in: /etc/apt/sources.list.d/unityhub.list
    1: deb https: //hub.unity3d.com/linux/repos/deb stable main
  Active apt repos in: /etc/apt/sources.list.d/vscode.list
    1: deb [arch=amd64,arm64,armhf] http: //packages.microsoft.com/repos/code stable main
Info:
  Processes: 350 Uptime: 17d 4h 50m Memory: 15.52 GiB used: 8.25 GiB (53.2%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.3.0 alt: 10/11/9 Client: Cinnamon v: 5.6.7 inxi: 3.3.13

VS Code:
Version: 1.75.0
Commit: e2816fe719a4026ffa1ee0189dc89bdfdbafb164
Date: 2023-02-01T15:29:17.766Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
OS: Linux x64 5.15.0-58-generic
Sandboxed: No

I am wondering if there is some place where I can get Platform.io logs to post for additional info?
I have let my laptop run all night at high CPU load to see if it ever got the project created, It didn't....
Platform.io version is: Core 6.1.6·Home 3.4.3
Any Ideas?

mrengineer7777, 2023-02-20:
What version of the VSCode PIO extension? 2.5.5 or 3.0.0?

CropWatchDevelopment, 2023-02-22:
Hello @mrengineer7777 I am using version 3.0.0

closed 2023-02-24 by ivankravets