platformio/platformio-vscode-ide #3496: ESPIDF framework bug

heliang1795, 2023-02-10:
in some examples   ;

 the defined  value  in  sdkconfig.defaults   was igroned

such as  :
sdkconfig.defaults:
CONFIG_MB_COMM_MODE_ASCII=y
CONFIG_MB_SLAVE_ADDR=1
CONFIG_MB_UART_BAUD_RATE=115200
CONFIG_FMB_TIMER_PORT_ENABLED=n
CONFIG_FMB_TIMER_ISR_IN_IRAM=y
CONFIG_FMB_TIMER_USE_ISR_DISPATCH_METHOD=y

 they were used in main.c  ,  but complier  did  not found ;

#define MB_PORT_NUM     (CONFIG_MB_UART_PORT_NUM)   // Number of UART port used for Modbus connection
#define MB_SLAVE_ADDR   (CONFIG_MB_SLAVE_ADDR)      // The address of device in Modbus network
#define MB_DEV_SPEED    (CONFIG_MB_UART_BAUD_RATE)  // The communication speed of the UART

mrengineer7777, 2023-02-20:
Wrong repository.  Please post on https://github.com/espressif/esp-idf.

ivankravets, 2023-02-24:
Please forward to https://community.platformio.org/

closed 2023-02-24 by ivankravets