platformio/platformio-vscode-ide #3549: Start PIO Home Server

C0bra071, 2023-02-21:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.75.1
  PIO IDE: v3.0.0
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
    at Timeout._onTimeout (c:\Users\Igor\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:182:9)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets

OmarBannour, 2024-07-12:
Error: Could not start PIO Home server: Timeout error
  at Timeout._onTimeout (c:\Users\Igor\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:182:9)
  at listOnTimeout (node:internal/timers:559:17)
  at processTimers (node:internal/timers:502:7)