platformio/platformio-vscode-ide #3566: Start PIO Home Server

wood-masmaster, 2023-02-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.75.1
  PIO IDE: v3.0.0
  System: Linux, 5.15.0-57-generic, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (/home/promaster/.vscode/extensions/platformio.platformio-ide-3.0.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:10392)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-02-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-02-24 by ivankravets

Lxl600, 2024-12-19:
打不开啊