platformio/platformio-vscode-ide #3588: Feature Request: variable holding the current env

Jason2866, 2023-02-27:
which can will allow users to use native VSCode's `workbench.action.tasks.runTask` and PlatformIO commands.

Example:
```
PlatformIO: Upload Filesystem Image (${currentEnvironment})
```

Reference https://github.com/platformio/platformio-vscode-ide/issues/2623#issuecomment-1447067634

Thx @ivankravets

ivankravets, 2023-02-28:
Fixed in https://github.com/platformio/platformio-vscode-ide/commit/27e1411255478143398af9161b0f8dc216b63642

Could you re-test with [PlatformIO IDE 3.0.1-beta.3](https://www.dropbox.com/s/tmaef56gvndey7n/platformio-ide-3.0.0-beta.3.vsix?dl=0) for VSCode?

Jason2866, 2023-02-28:
@ivankravets Nearly there :-) 

Works perfect if env is selected via Platformio VSC IDE Options.
[image]
when used Default in Platformio VSC IDE
[image]
and the default_envs is set in platformio.ini it does not work as expected
[image]

The window opens to select an env
[image]
an not the set default_env is used.

Jason2866, 2023-02-28:
Tested with
```
        "text": "$(refresh)",
        "tooltip": "Testing",
        "commands": [
          {
            "id": "platformio-ide.runPIOCoreCommand",
            "args": "pio run -t uploadfs -e ${command:platformio-ide.activeEnvironment}"
          }
        ]
      },
```
works fine with selection via Platformio VSC IDE.

Fails with set_default in platformio.ini since the `${command:platformio-ide.activeEnvironment}` is not set (undefined).

[image]

ivankravets, 2023-03-10:
Should be fixed in the pre-release version:

1. Navigate in VSCode to the Extensions activity bar
2. Click on the PlatformIO IDE
3. Switch to the PRE-RELEASE version.

Does it work now?

closed 2023-03-10 by ivankravets

Jason2866, 2023-03-10:
@ivankravets  Great, it does work now! Thx for your awesome work and support :-)