platformio/platformio-vscode-ide #3628: Start PIO Home Server

Joao-Done, 2023-03-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.76.0
  PIO IDE: v3.0.0
  System: Linux, 5.19.0-35-generic, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (/home/joao-victor/.vscode/extensions/platformio.platformio-ide-3.0.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:10392)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-03-10:
Could you reproduce this issue after the restart of the VSCode?

Joao-Done, 2023-03-10:
I'm sorry i have no recordings of posting this, and i'd like to close this thread thanks for the answer

closed 2023-03-10 by Joao-Done