platformio/platformio-vscode-ide #3634: Start PIO Home Server

KOE73, 2023-03-08:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.76.0
  PIO IDE: v3.0.0
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\oleg\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10392)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-03-10:
Could you reproduce this issue after the restart of the VSCode?

KOE73, 2023-03-10:
I completely reinstalled VS Code and PlatformIO. The problem was solved.

P.S. The problem was global, I reinstalled many programs.

Thank you.

closed 2023-03-10 by KOE73