platformio/platformio-vscode-ide #3643: Start PIO Home Server

Abdulrahman0044, 2023-03-10:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.76.0
  PIO IDE: v3.0.0
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\lenovo\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10392)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)
  ```

ivankravets, 2023-03-14:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3408

closed 2023-03-14 by ivankravets

Abdulrahman0044, 2023-03-20:
@ivankravets I didn't, it's working perfectly now