platformio/platformio-vscode-ide #3655: Last update ( Windows 7 )

Wiz-IO, 2023-03-15:
Hi there,

After last update .... ( https://github.com/platformio/platformio-vscode-ide/issues/3481 )
No PIO, No Home...  ( the old projects work )

[image]

ivankravets, 2023-03-16:
Could you provide the errors from VSCode: Help > Toggle Developer Tools > Console?

Wiz-IO, 2023-03-16:
[Desktop.zip](https://github.com/platformio/platformio-vscode-ide/files/10986701/Desktop.zip)

Hieromon, 2023-03-16:
A similar phenomenon occurs in Windows 10. But this phenomenon does not appear to occur in all Windows 10 environments. I have two different Windows 10 PCs and this issue occurs on one of them. The most difference is that the problem PC is joined to a Windows domain network and doesn't have full admin rights. It is always reproduced after updating to PIO-IDE 3.10.

```
PlatformIO Core version     6.1.6
Python                      3.9.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp932
PlatformIO Core Directory   C:\Users\hieromon\.platformio
PlatformIO Core Executable  C:\Users\hieromon\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\hieromon\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       4
Tools & Toolchains          17
```

Here are the steps to reproduce the issue:

1. Launch CSVODE with no folder specified.
2. Click PIO IDE extension icon.
3. `Initializing PlatformIO Core...` message responded.
4. But initialization of PlatformIO Core does not start.

Screenshot at this time is shown below.

[image]

Also, Here is the console log I got:

```log
workbench.desktop.main.js:82 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
onFirstListenerAdd @ workbench.desktop.main.js:82
_event._event @ workbench.desktop.main.js:82
(anonymous) @ workbench.desktop.main.js:80
onFirstListenerAdd @ workbench.desktop.main.js:80
_event._event @ workbench.desktop.main.js:82
B @ workbench.desktop.main.js:137
U @ workbench.desktop.main.js:542
createMouseController @ workbench.desktop.main.js:542
Q @ workbench.desktop.main.js:162
H @ workbench.desktop.main.js:542
z @ workbench.desktop.main.js:542
I @ workbench.desktop.main.js:543
createTree @ workbench.desktop.main.js:543
b @ workbench.desktop.main.js:543
ne @ workbench.desktop.main.js:701
ce @ workbench.desktop.main.js:2980
_createInstance @ workbench.desktop.main.js:599
createInstance @ workbench.desktop.main.js:599
createTree @ workbench.desktop.main.js:2980
activate @ workbench.desktop.main.js:2980
setVisibility @ workbench.desktop.main.js:2980
updateTreeVisibility @ workbench.desktop.main.js:2980
(anonymous) @ workbench.desktop.main.js:2980
invoke @ workbench.desktop.main.js:82
deliver @ workbench.desktop.main.js:82
fire @ workbench.desktop.main.js:82
setVisible @ workbench.desktop.main.js:2089
(anonymous) @ workbench.desktop.main.js:2980
setVisible @ workbench.desktop.main.js:2980
setVisible @ workbench.desktop.main.js:2980
showComposite @ workbench.desktop.main.js:2089
doOpenComposite @ workbench.desktop.main.js:2089
openComposite @ workbench.desktop.main.js:2089
doOpenViewlet @ workbench.desktop.main.js:3058
openPaneComposite @ workbench.desktop.main.js:3058
run @ workbench.desktop.main.js:2764
runAction @ workbench.desktop.main.js:82
run @ workbench.desktop.main.js:82
onClick @ workbench.desktop.main.js:176
(anonymous) @ workbench.desktop.main.js:176
Show 10 more frames
DevTools failed to load source map: Could not load content for https://ticino.blob.core.windows.net/sourcemaps/da76f93349a72022ca4670c1b84860304616aaa2/core/vs/workbench/workbench.desktop.main.js.map: Load canceled due to load timeout
```

NOTE:

If I open a folder that is already configured as a PIO project, PIO is initialized without problems and I can execute PIO tasks such as builds or upload.

ivankravets, 2023-03-16:
@Wiz-IO , I don't see any problems with the logs. Everything is loaded well. Maybe you disabled PlatformIO IDE Toolbar? Could you "right click" on the blue (status bar) and check if the toolbar is enabled?

ivankravets, 2023-03-16:
@Hieromon , could you try to disable ALL extensions but keep enabling PlatformIO IDE + C/C++ from Microsoft? Restart VSCode. Does it work now?

Wiz-IO, 2023-03-16:
look
https://www.youtube.com/watch?v=C3ukQzWJQMM

ivankravets, 2023-03-16:
I see where is a problem. Old VSCode....

Please try the [3.1.1 Beta.3](https://www.dropbox.com/s/ynk41anb0jwfpko/platformio-ide-3.1.1-beta.3.vsix?dl=1). Does it work now?

Wiz-IO, 2023-03-16:
[VSIXInstaller.txt](https://github.com/platformio/platformio-vscode-ide/files/10994801/VSIXInstaller.txt)

Hieromon, 2023-03-17:
@ivankravets, Verified on Win7 and Win10 which I can test right now. As you can see, the issue has been resolved. There is no interference with other extensions in any of the below cases.

With the combination of PIO-IDE 3.1.0 and VSCode 1.76.0, I can reproduce this problem in Win7 as well as Win10. The problem does not occur with PIO-IDE 3.1.1-Beta3.

On the other hand, VSCode 1.76.2 does not cause any problems even with PIO-IDE 3.1.0.

Thanks for your great support.

ivankravets, 2023-03-17:
Resolved in https://github.com/platformio/platformio-vscode-ide/commit/2b97f63b2b3da0c21d83c18059855906d4533c5d

The official PlatformIO IDE 3.1.1 for VSCode is out! Please update!

closed 2023-03-17 by ivankravets