platformio/platformio-vscode-ide #3656: Regression bug - v3.1.0 requires interaction to restart monitor

birnam, 2023-03-15:
In the latest version v3.1.0, clicking "Upload and Monitor" after it is already monitoring causes a dialog box to terminate request or restart process. This was previously reported in https://github.com/platformio/platformio-vscode-ide/issues/2266 .

The regression appears to be due to the work in https://github.com/platformio/platformio-vscode-ide/issues/2623 (I had originally reported this in that closed issue).

[image]

ivankravets, 2023-03-16:
Thanks for the report. Could you re-test 3.1.1-beta.1 version?

1. [Download VSIX](https://www.dropbox.com/s/zl243gang09zkxc/platformio-ide-3.1.1-beta.1.vsix?dl=1)
2. Navigate to VSCode > Extensions > `...` > Install from VSIX...

Does it work now as expected?

closed 2023-03-16 by ivankravets

ivankravets, 2023-03-16:
@matteocarnelos , added in https://github.com/platformio/platformio-vscode-ide/commit/3a43a12f510483776b2cb7b6deeb8740e93ee988

The problem is linked to the limited VSCode API. They do not provide any instruments to handle "Before Task Start".

Could you try the [3.1.1 Beta 2](https://www.dropbox.com/s/pkhc2w6ea6skoee/platformio-ide-3.1.1-beta.2.vsix?dl=1) and ``platformio-ide.uploadAndMonitor``?

matteocarnelos, 2023-03-16:
Tested it in Beta 2 and it works! Thanks for the quick fix!

ivankravets, 2023-03-17:
The official PlatformIO IDE 3.1.1 for VSCode is out! Please update!

jareklupinski, 2023-03-24:
I am on 3.1.1 and this is still happening to me:

[image]

[image]

matteocarnelos, 2023-03-24:
@jareklupinski what's your `platfromio-ide.toolbar` entry?

Mind that the fix applied in `3a43a12` only works using the `platformio-ide.uploadAndMonitor` task.

Therefore, if you have the following entry in your `platformio-ide.toolbar` array:
```json
{
    "text": "$(arrow-right)",
    "tooltip": "PlatformIO: Upload & Monitor",
    "commands": [
        {
            "id": "workbench.action.tasks.runTask",
            "args": "PlatformIO: Upload and Monitor (${command:platformio-ide.activeEnvironment})"
        }
    ]
}
```

Consider changing it to the following:
```json
{
    "text": "$(arrow-right)",
    "tooltip": "PlatformIO: Upload and Monitor",
    "commands": "platformio-ide.uploadAndMonitor"
}
```

Note that the "Upload and Monitor" task will be run on the default environment.

jareklupinski, 2023-03-24:
Thanks @matteocarnelos , this is how it looked:

[image]

I changed the lines as you said, restarted the IDE, then tried pressing the new Upload and Monitor button on the bottom-left. The process completed, but when I clicked it again, I got the same popup:

[image]

matteocarnelos, 2023-03-24:
@jareklupinski the second screenshot you sent still contains the "old" configuration.

Do the following:
* Open `settings.json`
* Replace the "old" configuration with the following:
  ```json
  {
      "text": "$(arrow-right)",
      "tooltip": "PlatformIO: Upload and Monitor",
      "commands": "platformio-ide.uploadAndMonitor"
  }
  ```
* Save the file
* Restart VSCode
* Launch the task from the toolbar
* Re-launch the task and see if the problem persists

If this doesn't work I don't really know how to help.

jareklupinski, 2023-03-25:
@matteocarnelos I tried that as well, saved and restarted VSCode before trying again:

[image]

matteocarnelos, 2023-03-25:
@jareklupinski unfortunately I don't know how to help at this point.

You might want to open a new issue to keep track of this strange behavior.

commit 0f7a505a (2023-03-16): Fixed a regression bug where running the same "monitor" causes the notification that the task "is already active" // Resolve #3656