platformio/platformio-vscode-ide #3657: Unity tests aren't uploaded if several ESP32's is connected

nicklasb, 2023-03-15:
Hi everyone!

Thanks for a great tool!

I have this odd issue. I have to disconnect ESP32s for unity tests to upload. 
**Situation:**
1. I have several ESP32's of different types (and different ports) connected in the same project.
2. They all have their separate upload ports defined in their separate environments.
3. They are connected through a USB hub.
4. I run Unity unit tests on them all. 
5. I have a Mac. For now.
6. "Upload and monitor" works as expected. 

**The problem**
I have to disconnect all ESP32s that I am not currently testing for the code to upload and the new testing to run. 
If I don't, it _seems_ like the tests upload, and the tests run. But it is still the old tests.
I also have an Arduino Uno connected. 
That, or any other USB devices doesn't have to be disconnected for some reason, just the ESP32s.

I don't get it, is there some kind of limitation I don't know of, or have I just not understood something? 
If so, why does the regular "upload and monitor" work?

ivankravets, 2023-03-16:
Have you tried to define [upload_port](https://docs.platformio.org/en/latest/projectconf/sections/env/options/upload/upload_port.html) and [test_port](https://docs.platformio.org/en/latest/projectconf/sections/env/options/test/test_port.html)?

nicklasb, 2023-03-16:
Ouch! That was it, had completely missed that setting, thank you! 
Closing!

closed 2023-03-16 by nicklasb