platformio/platformio-vscode-ide #3658: Start PIO Home Server

copper1999, 2023-03-16:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.76.2
  PIO IDE: v3.1.0
  System: Darwin, 22.1.0, arm64

  %23 Exception
  ```
  Error: Error: spawn /Users/copper/.platformio/penv/bin/python ENOENT
	at /Users/copper/.vscode/extensions/platformio.platformio-ide-3.1.0-darwin-arm64/node_modules/platformio-node-helpers/dist/index.js:1:10645
	at o (/Users/copper/.vscode/extensions/platformio.platformio-ide-3.1.0-darwin-arm64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.<anonymous> (/Users/copper/.vscode/extensions/platformio.platformio-ide-3.1.0-darwin-arm64/node_modules/platformio-node-helpers/dist/index.js:1:5161)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess._handle.onexit (node:internal/child_process:289:12)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)
  ```

ivankravets, 2023-03-17:
Could you reproduce this issue when restarting VSCode?

ivankravets, 2023-05-10:
Please remove `/Users/copper/.platformio/penv` folder and restart VSCode.

closed 2023-05-10 by ivankravets